rusefi/rusefi-hardware #7: PDM

rusefillc, 2021-11-17:
fans/starter
to replace relay box, this both weight reduction and monitoring
current per device

mmatousek89, 2021-11-17:
https://www.ecumaster.com/products/pmu/

mmatousek89, 2021-11-17:
http://www.tremondi.cz/motorsport/power-units/

mmatousek89, 2021-11-17:
https://www.infineon.com/cms/en/product/power/smart-low-side-high-side-switches/high-side-switches/

mmatousek89, 2021-11-17:
https://www.infineon.com/cms/en/product/power/smart-low-side-high-side-switches/high-side-switches/profet-plus-2-12v-automotive-smart-high-side-switch/bts70015-1esp/

mmatousek89, 2021-11-17:
https://www.st.com/resource/en/datasheet/cd00002198.pdf

mmatousek89, 2021-11-17:
https://github.com/UVicFormulaMotorsport/PDU

rusefillc, 2021-11-18:
With availability drama do we like https://www.rohm.com/products/power-management/ipds/ ?

rusefillc, 2021-11-18:
Let me ask the guy if this runs stm32 or not

[image]

rusefillc, 2021-11-18:
Dave B.  6:08 PM
the mosfet vs. protected driver issue is an interesting one
6:08
if you're going to do digital current monitoring, you could probably get away with regular mosfets as long as you have a fast control loop for monitoring current and shutting down on a overcurrent event, and some kind of TVS clamp for inductive loads.  Wouldn't cut it for PWM/frequency control due to thermals but for solenoids that were infrequently toggled it would be fine. (edited) 

6:09
PWM outputs get a recirculating flyback diode
6:09
or half bridge / FET as recirc diode
6:10
you could totally do a discretes PDM with a bunch of sense resistors and current sense amps
6:11
hell, you could probably use the RDSon of the FETs as the current sense resistor.  if you added a thermistor for each FET and did temperature compensation, you could probably get close enough (tm) with temperature compensated current sensing

rusefillc, 2021-11-19:
https://vk.com/tdgelectronics used AUIR3313S with stm32f405rgt