rusefi/rusefi-hardware #9: GDI board in 121 case

rusefillc, 2021-12-29:
We have proven GDI board which requires SPI initialization and takes TTL control signals https://github.com/rusefi/rusefi/tree/master/hardware/Common_Rail_MC33816

We need an integrated unit with CAN some ARM processor (looks like GD32F130) and low side control signal option, pin compatible with BMW HDEV unit

it would be nice to increase percentage of JLC assembly

[image]

rusefillc, 2022-01-04:
See https://github.com/rusefi/rusefi/wiki/Passat-GDI-wiring

note 3v jumper which is only visible on the last image!!!

rusefillc, 2022-01-04:
HDEV pinout
https://raw.githubusercontent.com/wiki/rusefi/rusefi/OEM-Docs/Bmw/2003_7_Series_e65/2003_N73_engine_2.png
https://raw.githubusercontent.com/wiki/rusefi/rusefi/OEM-Docs/Bmw/2003_7_Series_e65/2003_N73_engine_3.png

rusefillc, 2022-01-04:
Let keep in mind expending this board to https://github.com/rusefi/rusefi/tree/master/hardware/Common_Rail_MC33816 later with two additional injector channels

rusefillc, 2022-01-04:
@Qwerty-OFF I forgot to mention - we have https://github.com/rusefi/rusefi-hardware/tree/main/mini48-stm32 and https://github.com/rusefi/rusefi-hardware/tree/main/wideband-F103 here with 48 pin stm32 footprint and even 12v power supply/CAN

rusefillc, 2022-01-04:
once we have the schematics and see what pins are available we will probably throw a few aux analog inputs and maybe even some outputs on the board

rusefillc, 2022-01-08:
pin 51: "ignition" - that's +12v power supply for MCU regulator / MCU power
pin 52 yellow/black is CAN-H
pin 53yellow/brown is CAN_L

on 4 channel board let's NOT use BMW pins for injectors 5 and 6
pins 47 and 47 are just vias
let's use pin 40 for pump control
let's use 82+83 for pump output LOW and 84+85 for pump output HIGH

ECU X60203.51 yellow/brown: CAN_L
ECU X60203.38 yellow/black: CAN_H

rusefillc, 2022-02-22:
pin 51 clarification:

we are trying to be as compatible with https://github.com/rusefi/rusefi/wiki/OEM-Docs/Bmw/2003_7_Series_e65/2003_N73_engine_1.png https://github.com/rusefi/rusefi/wiki/OEM-Docs/Bmw/2003_7_Series_e65/2003_N73_engine_2.png as possible

it looks like OEM pin 51 "pin 15" goes from ECU?
while pin 4 goes from main relay?

Qwerty-OFF, 2022-02-22:
[image]
IRQB, DRVEN, RESETB, FLAG0-3 - any free pin of stm32?
OA1-2 - ADC input of stm32?
START1-START6 directly from main connector? How to drive pump by only one pin 40?

mck1117, 2022-02-22:
For STARTx inputs:

[image]

This will allow us to drive STARTx from a normal lowside output on an ECU.

closed 2022-03-20 by Qwerty-OFF