rusefi/rusefi-hardware #19: WBO-F103: silkscreen

rusefillc, 2022-01-10:
We need silkscreen for all pins. Looks like back side would be perfect?

while we are on that subject we can use some board name on the back maybe even with some URL maybe even https://github.com/s/wb

rusefillc, 2022-01-13:
@blundar I am not sure if I am doing anything wrong but I do no see silk on back of PCB :(

[image]

can you plz paste a screenshot of how it looks on your end?

dron0gus, 2022-04-10:
[image]
How it looks like after https://github.com/rusefi/rusefi-hardware/pull/60 .

closed 2022-04-10 by dron0gus