rusefi/rusefi-hardware #26: MRE V2

OrchardPerformance, 2022-01-15:
With the availability of TLE8888 chips becoming bad it looks like the old version of the MRE is showing it's age and has issues that we could address with a V2. 

Problems: 
Heat of the power stage
Availability and cost of TLE8888 
Useless flash chip  
Lack of IO for some uses 
Issues with main plug pinout resulting in signal noise
Components stuffed between pins on main plug
More 6 cylinders that we expected asking about MRE, also less bikes and sub-3 cylinder engines 
Hall and VR needs soldering to change 
No Knock sensor circuit 
No room for external waterproof USB 
No WBO2 option 
MRE has potential to short 12v to 5v rail with incorrect JP2 soldering

Solution: 
Boshun BS603 case has 64 pin plug and is only 35mm longer than existing case. 
64 pin plug means we can resolve the pinout problems
Use extra 16 pins to have dedicated Hall/VR input with new discrete VR circuit. 
Use extra pins to add 2x push pull and 2x lowside outputs 
Extra PCB space prevents components between main plug pins
Upgrade the MRE to a switching power stage like the Proteus 
Replace TLE8888 with discrete components 
Replace flash with SD card (thanks to JLD) 
Add knock sensor circuit from Proteus. 
Introduce a "keep out" zone on the board for extra room to fit waterproof external USB
Add WBO2 circuit 
Use extra room to make JP2 a DPST switch

Obvious issues: 
Dev time - If we have to solve watchdog or source TLE8888 that is also dev time. 
Existing cases - We can use the 48 pin case for the body module that has been discussed, we can still use them
"I want a small engine ECU" - Just use an MRE anyway or the AlphaX really small engine ECU, I can think of only one bike install and nothing less that 3 cylinders anyway. 
Cross over with Proteus - Not really an issue, MRE is in my experience the first choice of most people and Protues only when they run out of IO, Proteus going to Atlas changes things anyway and means we could probably retire Proteus.

Obvious benefits: 
Makes MRE V2 and Atlas only range viable. 
Covers all basic engines with the MRE (up to 6 cylinders, Wasted+Batch V8) 
Aligns MRE with the full 2022+ rusEFI software capabilities 
64 pin plug means improved IO, better routing and no chance of accidental plug in with an MRE V1 
Cost of unit will be roughly the same as existing MRE or less, 64pin case is the same price, TLE8888 cost gone. 
Slightly expanded features make the slightly higher prices we have had to set much more reasonable

OrchardPerformance, 2022-01-15:
@mck1117 @rusefillc 
Tagging you guys for discussion

rusefillc, 2022-01-15:
I am super tired of hardware design and I have a queue of plug-in boards to design. Somewhere in the next couple of months I will attempt to order a new batch of MRE since that's the path of least resistance. For cool ECU we have proteus. Boards based on Hellen ECU have a bit of extra size let's call it "Hellen tax". Let's see how Hellen Harley goes.

https://github.com/rusefi/alphax-4chan would be a universal smaller ECU but with only 48 pins it would be somewhat limited and not an improvement over MRE.

OrchardPerformance, 2022-01-15:
This would certainly not be near future. 
Longer term strategy thing to align features.

rusefillc, 2022-01-15:
History says we are not great at long term planning

rusefillc, 2022-07-25:
this seems to describe https://github.com/rusefi/alphax-4chan/ and/or https://github.com/rusefi/alphax-8chan/ which are work in progress

closed 2022-07-25 by rusefillc