rusefi/rusefi-hardware #33: pump channel control

rusefillc, 2022-02-23:
We are confused about fuel pump solenoid control

Which of start5 and/or start6 is controlling pump?
Does microcode match existing board?

Is pump controlled by both start5 and start6? if it's not, what is the other one controlling?

what pin do we expose on GDI4 connector with what kind of electrical considerations?

[image]

[image]

NOx-z, 2022-02-23:
per line 48 on the source on file,
https://github.com/rusefi/rusefi/blob/master/firmware/hw_layer/mc33816/rusefi/MicrocodeCh2/ch2.psc#L48
We use START6, and we turn on High-Side5 (VBat), and Low-Side6 drivers with it.

At a glance the board appears accurate. There is no populated LowSide5.

We made the decision to only map one (excluding START5), and left the other unmapped + unpopulated because the only way we could use the other was to then switch off whatever START6 was targeting, they both can not be on at the same time. We examined general use cases and found we did not have another solenoid to power intermittently.

I am unaware at the moment where the pins for GDI4 are defined. All of the START signal should be similar, that is, they are enabled at a high level (VccIO) of the MC33816/PT2001. They are not driving anything themselves beyond a signal.

rusefillc, 2022-02-26:
@Qwerty-OFF let's do both "same as injectors" via external pin and a jumpers allowing to switch to internal pump channel control from stm32f103

rusefillc, 2022-03-21:
What jumper controls internal/external pump control?

please add explicit comment right on schematics

Qwerty-OFF, 2022-03-21:
No you can use wire frome any stm32 pad:
[image]
to connector jumpers:
[image]
Do we need special jumper for this?

rusefillc, 2022-03-21:
Yes lets add a 0R

Qwerty-OFF, 2022-03-21:
[image]

closed 2022-03-21 by Qwerty-OFF