rusefi/rusefi-hardware #34: digital input board firmware

rusefillc, 2022-02-26:
we have some hardware to test at least analog muxing

need firmware

andreika-git, 2022-02-26:
> with nucleo detached from green board PF3 jumped to 3v directly reports around 3100 while I would expect much closer to 4095
> with nucleo detached from green board PF3 jumped to 3v directly reports around 420 while I would expect much closer to 0?

Looks like an "ADC echo" effect when measuring on un-connected (or anassigned) ADC input somewhere near the correct one...

abelom, 2022-03-24:
@dron0gus has fixed ADC - I have proper working ADC without green board now see https://github.com/rusefi/rusefi-hardware/commit/9add16109be7810699296a0d0019e1342f563564

see potentiometer soldered between TST1 and +12v

@Qwerty-OFF need help with test scenario: please give me some specific ADR and OUT set of values with some specific test TST test condition to read any specific voltage on IN0.

Qwerty-OFF, 2022-03-24:
Set:
A0 = A1 = A2 = A3 = 0
OUT1 = 1
TST17 = open
Rezult:
IN1 = 3.3V

Set:
A0 = A1 = A2 = A3 = 0
OUT1 = 0
TST17 = open
Rezult:
IN1 = 0V

Set:
A0 = A1 = A2 = A3 = 0
OUT1 = any
TST17 = short to GND
Rezult:
IN1 = 0V

Set:
A0 = A1 = A2 = A3 = 0
OUT1 = any
TST17 = short to +12
Rezult:
IN1 = 3.3V

rusefillc, 2022-03-25:
blocked by #50

rusefillc, 2022-06-25:
@Qwerty-OFF 

https://youtu.be/UVUBTyf_aaM

не-делённое напряжение соответсвенно тихонько падает с 10ти вольт - это подтверждено мультиметром

rusefillc, 2022-06-27:
@Qwerty-OFF says

Не могу понять полную картину управления, но это пока кажется не главное. ADR0-ADR3 и MUX_OFF инверсные, чего я почему-то не написал раньше, т.е. чтобы поставить желанный второй контакт нужно выставить ADR0 в 0, остальные в 1, и MUX_OFF в 1

rusefillc, 2022-06-27:
progress!!!

```

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
0: 2.868 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
1: 0.108 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
0: 2.870 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
1: 0.004 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
0: 2.783 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
1: 0.004 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
0: 2.869 mV

```

rusefillc, 2022-06-27:
цикл от 0 до 15

```
Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
0: 2.902 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
1: 0.005 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
2: 2.872 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
3: 0.004 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
4: 2.872 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
5: 0.037 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
6: 2.903 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
7: 0.041 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
8: 2.904 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
9: 0.002 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
10: 2.872 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
11: 0.004 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
12: 2.872 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
13: 0.208 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
14: 2.905 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
15: 0.095 mV

Hello

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
0: 2.902 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
1: 0.002 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
2: 2.871 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
3: 0.004 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
4: 2.871 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
5: 0.037 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
6: 2.904 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
7: 0.038 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
8: 2.902 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
9: 0.004 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
10: 2.871 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
11: 0.004 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
12: 2.843 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
13: 0.038 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
14: 2.904 mV

output 1: bit=0 0
output 1: bit=1 1
output 1: bit=2 1
output 1: bit=3 1
15: 0.037 mV
```

rusefillc, 2022-06-27:
@andreika-git we have test data! Does it match https://github.com/rusefi/stim/issues/6#issuecomment-877653918 expectation?