rusefi/rusefi-hardware #46: START5/PUMP1

rusefillc, 2022-03-21:
we don't actively use START5/PUMP1 .. what if we make space on the PCB for the required components to be manually populated later? just in case

rusefillc, 2022-03-21:
@nox-z please elaborate a bit? what components? any reference?

NOx-z, 2022-03-21:
replicating as to how the injector banks allow 2 different lowsides. the Q12, C63, R20, D16, C67 would be replicated/duplicated, and this fet output would tie into the sense resistor, R26.
Also, the current label for the single pump circuit out is `L5`, this should be `L6` if the D/G_LS6 are used. Recommendation to change the label of the output on the connector to be N/P (Negative/Positive) like the injectors instead of L/H (Low/High), for consistency.

rusefillc, 2022-03-21:
Low priority, only if not a major pain in terms of space.

Qwerty-OFF, 2022-03-21:
L5\L6 fixed
At now we have place only for HS5 or LS5. I can change HS5 to LS5. Placement of HS5 and LS5 together will require significant changes.

rusefillc, 2022-03-21:
Please stay as compatible with existing board as possible to use same pt2001 microcode

Qwerty-OFF, 2022-03-21:
OK, I keep Common_Rail_MC33816 structure with HS5

closed 2022-04-07 by rusefillc