rusefi/rusefi-hardware #50: digital input board firmware: ADDR gpio bug

rusefillc, 2022-03-25:


rusefillc, 2022-03-25:
PC5 I see a HW conflict

[image]

rusefillc, 2022-03-25:
@dron0gus need help :(

dron0gus, 2022-03-26:
[image]
Removing SB177 should put ethernet Phy into reset. This should keep all phy's outputs in hi-z state.

mck1117, 2022-03-26:
You could also hold that pin low with the stm32 if it's connected to a gpio.

rusefillc, 2022-03-26:
Does this explain pc8 drama?

rusefillc, 2022-03-26:
As for ETH i was thinking to simply remove stuff from nucleo

rusefillc, 2022-03-31:
https://www.st.com/resource/en/user_manual/dm00244518-stm32-nucleo-144-boards-stmicroelectronics.pdf says that PC5 has SB181

[image]

rusefillc, 2022-03-31:
With SB181 removed I am at

ADR0=PC8: 3.3v
ADR1=PC6: 3.3v
ADR2=PC5: 3.3v

dron0gus, 2022-03-31:
https://github.com/rusefi/rusefi-hardware/pull/52