rusefi/rusefi-hardware #55: 9779 review

rusefillc, 2022-03-31:
We have https://github.com/rusefi/rusefi-hardware/tree/main/L9779WD-breakout in kicad5 MRE name

open question: what shall we do with

[image]

dron0gus, 2022-04-06:
OUT_21..OUT24 to OUTA..OUTD of 9779.
FLCP -> SCK of 9779.
Looks like INJ_IN_* and INJ_* are swapped on 9779. INJ_IN_* are logic level signals from STM32 should go to 9779's IN1..IN4.
IN_9..IN_12 nets to IN5..PWM/IN8 inputs of 9779.
OUT_5, OUT_6, OUT_7 to OUT5, OUT6, OUT7 of 9779.

dron0gus, 2022-04-06:
@rusefillc do we want to replace two discrete ignition drivers (U32, U33 MIC4427) with L9779's integrated ignition pre-drivers?

dron0gus, 2022-04-06:
[image]
Delete.

dron0gus, 2022-04-06:
Add 0R jumpers (not populate by default) between following nets:
OUT_14 - AV9
OUT_15 - AV8
OUT_16 - AV10
OUT_17 - AV6
Mapping can be adjusted for easy routing.

dron0gus, 2022-04-06:
9779.OUT_VRS -> CRNK_IN

rusefillc, 2022-04-06:
@dron0gus yes let's use 9779 features to the most

dron0gus, 2022-04-06:
@rusefillc ok!
R21, R33, R80, R81 - replace with 100R. Mark populated.
J51, J52, J53, J54 - remove
U32, U33, R122, R123, r124, R125, C78, RN5 - remove.
Ign1_MCU..Ign4_MCU nets - remove.

hasl58, 2022-04-13:
[image]

?

hasl58, 2022-04-13:
[image]
?

dron0gus, 2022-04-14:
OUT_19 OUT_20 are not used outside this sheet, so can be remover.
[image]

dron0gus, 2022-04-14:
As LowSides_1-2 and LowSides_3-4 are now deleted, DFB10 .. DFB 14 and OUT10 .. OUT14 are now no longer used - delete.
OUT8..OUT9 and DFB8..DFB9 are also not used - delete.
WDA_INT to some MCU gpio (if any free left).

closed 2022-07-18 by rusefillc