rusefi/rusefi-hardware #91: GDI4 did I blow F1?

rusefillc, 2022-06-07:
my F1 no longer beeps continuity
MCU and PT2001 are alive if I power board with 5v

car starts on BMW injector box!
GDI4 was happy with same harness on bench

first attempt to start real - now I have no LEDs on GDI board "assembled by Matt"

F1 fuse https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C310989 does not beep in DMM continuity test

Q1: what did I do wrong? :)
Q2: need replacement part of F1 ideally from mouser, digikey would also work

mck1117, 2022-06-07:
Under what circumstances did the fuse pop?

rusefillc, 2022-06-07:
I know that stitty 1.5A powe supply cannot even start gdi4 in one go

Takes two attempts to start the board!

That points to maybe inrush filling the capacitors
Car battery has no such problem

rusefillc, 2022-06-08:
big external 10A does not blow but #92

[image]

mck1117, 2022-06-13:
This may actually be fixed by #101 - without current sense the boost converter could be doing stupid things to the power supply.

rusefillc, 2022-06-15:
Yes

rusefillc, 2022-06-19:
solved by fixing shunt soldering

i run with 5A SMD fuses borrowed from Harley hellen

closed 2022-06-19 by rusefillc