rusefi/rusefi-hardware #95: huge holes/vias for PT2001 main pad

rusefillc, 2022-06-10:
can we do similar to microRusEFI maybe with 1mm diameter holes?

[image]

rusefillc, 2022-06-10:
Why did Q2 fry? What was I doing? well, while the board was powered, I believe I was moving the test injector from one injector output to another.. also my 'bench' isn't exactly stable - I just have wired for the injector->outputs just twisted together, so it can easily happen that it disconnects mid-run.. either during idle, or peak, or hold... is it reasonable to say this will happen inside of a car with many vibrations? or just bad user unplugging one mid-run? And better yet: how do we protect against Q2 blowing, since it also happened to me on R0.3 boards

[1:20](https://rusefi.slack.com/archives/CFP3QAJCW/p1654881630526919)
also, comparing the eval kit schematic to ours, the only real difference I can see is a small RC before the 10uH inductor (that could help with our noise?) .. so I dont particularly see much difference.. but they may not have seen this issue on the bench neither

Qwerty-OFF, 2022-06-11:
[image]

closed 2022-06-11 by Qwerty-OFF