rusefi/rusefi-hardware #98: GDI4 revisit D10

rusefillc, 2022-06-11:
@NOx-z says "while comparing datasheets from current parts vs legacy, D10 - the old part SBR10200CT is a 200v diode vs the new MBRD10100CT is 100v - which is worth having a head note as we brush up against that 100v rail"

@mck1117 @NOx-z @Qwerty-OFF is this OK or not OK?

Qwerty-OFF, 2022-06-11:
I see MBRD10100CTTR in last BOM version of CommonRail board
[image]

mck1117, 2022-06-14:
I'm actually okay with the ratings on that part - I see no real issue. That diode only sees ~75v max with our 65v rail, and that's enough headroom for me to be happy.

closed 2022-07-18 by rusefillc