rusefi/rusefi-hardware #99: Let's spell out all differences between GDI boards

rusefillc, 2022-06-11:
PT200x need firmware and hardware to work together for proper results, unfortunatelly boring details seem to matter

@Qwerty-OFF please write a report on BOM and schematics differences between GDI4 rev A and https://github.com/rusefi/rusefi/tree/master/hardware/Common_Rail_MC33816

rusefillc, 2022-06-11:
See https://github.com/rusefi/rusefi-hardware/issues/98
See https://github.com/rusefi/rusefi/issues/4251

Qwerty-OFF, 2022-06-11:
| Description  | GDI4 rev. A part  |  CommonRail part |
|---|---|---|
| All n-channel MOSFETs  | NCE0140KA_C194346  | IRLR3110ZTRPBF_C67277  |
| Boost inductor  |  BCIHP1770-100M_C237980 | IHLP6767GZER100M01 (not sure, non-JLC)   |
| Filter inductor  | MS1350-6R8M_C380073  | 744710610 (not sure, non-JLC)  |
| Output diodes | MBRD10100CTTR_C135494 | STTH802G-TR_C181034 | 

Large electrolytic capacitors have been replaced with several small capacitors
Also possibly changed cases of small parts from 0805 to 0603

mck1117, 2022-06-11:
What's the ESR difference on the caps?

Did sense resistor values change?

Qwerty-OFF, 2022-06-11:
Old caps: UHE2A391MHD (390 uF), SLPX102M100A3P3 (1000 uH)
New caps: VKMI1701K391MV_C487421

Sense resistors not changed

rusefillc, 2022-06-11:
@Qwerty-OFF see linked tickst on 10m vs 15m

Qwerty-OFF, 2022-06-11:
I see this in CommonRail BOM:
[image]
All sense resistors is 10mOhm

mck1117, 2022-06-12:
I remember now - we standardized on 10mOhm on the last rev of the old board. Let's stay with 10m on all.

mck1117, 2022-06-13:
Fixed by #101

closed 2022-06-13 by mck1117