rusefi/rusefi-hardware #101: split resistor pads are very error-prone

rusefillc, 2022-06-13:
[image]

note how resistors flow to one side and leave second part of the pad disconnected!

what is this evil pattern of split pad?! why and how?!

[image]

rusefillc, 2022-06-13:
Affected parts:

R21 R22 R22 R26

Qwerty-OFF, 2022-06-15:
On the CommonRail board, the shunts were connected incorrectly. A separate conductor must go from the GND pad of the shunt to the chip. As you can see, I made a separate pin for this, but the resistor moved to a larger pad when soldering.

Now the measuring contact is in the middle and the resistor should not move.

[image]

mck1117, 2022-06-15:
That still has potential for mfg errors. Why not just route a trace to the inside of the resistor?

rusefillc, 2022-06-15:
For mfg we want two pads of copper, not three not 6.

@Qwerty-OFF point is that kicad would be very confused in terms of DRC if coppers are connected, effectively being connected to GND

I wonder if we need to learn how to kicad this

mck1117, 2022-06-15:
DRC actually doesn't care. You can have two touching pads no problem. It comes out as a single piece of copper.

Qwerty-OFF, 2022-06-15:
Figured out how to get KiCAD to connect two pads with different nets in the same footprint without DRC issues. We can applaud the developers of KiCAD for having a special keyword for this.
[image]

Now definitely no soldering problems

commit 729573ab (2022-06-15): Fixed #101
commit fcce04b6 (2022-06-15): Second fix #101
closed 2022-06-15 by Qwerty-OFF