rusefi/rusefi-hardware #102: PT2000 board to have a HW mode to drive 8 injectors

rusefillc, 2022-06-14:
technically PT2000 can drive x8 of injectors instead of driving fuel pump

let's have flexibility via HW jumpers if jumpers are needed

rusefillc, 2022-06-14:
open question of x8 injectors should be the default HW config

Qwerty-OFF, 2022-06-15:
Can not. There are only two drivers in the pump bank: High Side and Low Side. The bank of injectors contains four: two High Side and two Low Side.
[image]

rusefillc, 2022-06-15:
@NOx-z ?

mck1117, 2022-06-15:
IMO the 6ch is a mistake in the first place.

But. That again seems like unnecessary complexity - why do we want to make our lives more difficult doing that? Why not run two 4ch boards on a v8?

rusefillc, 2022-06-15:
Pt2001 board is PnP into 760 test mule harness

mck1117, 2022-06-15:
> Pt2000 board is PnP into 760 test mule harness

sure - seems like we should do 100/200/500 miles on VW 4cyl before spinning another very similar board that will have the same problems we detect (if any) on the 4ch.

NOx-z, 2022-06-15:
Any High-Side driver can be used as Low-Side. This is true for the PT2001/MC33816 as well.
The PT2000 mentions this in the datasheet at (page 41 in pdf):
`6.3.3 High-side pre-drivers in low-side configuration`
and all drivers are accessible by any channel/core, but there are some special situations where only the dc-dc boost is applicable.

a configuration of 2 banks would include using 2 high side drivers per, as high side drivers (typical vboost and vbat) , and then a total of 4 more drivers (mix of high side + low side).

the PT2001 can use the 'fuel pump' drivers in order to drive 6 total cylinders, and no 'pump'/fuel limiting valve'

rusefillc, 2022-06-15:
See also #106

rusefillc, 2022-06-15:
@mck1117 i can agree to one mile. I am a believer in separation of responsibilities, once it did a mile i would need others to step in

mck1117, 2022-06-15:
seeing as nobody else owns a GDI vehicle they're willing to ecu swap - that separation of responsibilities (as you call it) would require you buying me a car

rusefillc, 2022-06-15:
I like that plan!

rusefillc, 2024-10-04:
> sure - seems like we should do 100/200/500 miles on VW 4cyl before spinning another very similar board that will have the same problems we detect (if any) on the 4ch.

@ElDominio I know you've made a dyno video about GDI Hyundai but do you know if any of those have put any miles?