rusefi/rusefi-hardware #106: PT2001 board to run 6 injectors

rusefillc, 2022-06-15:
Can a pt2001 board run six equal injectors without pump control?

mck1117, 2022-06-15:
but why?

rusefillc, 2022-06-15:
E66 760

NOx-z, 2022-06-30:
my first suggestion, before really looking over it, is to rename it as `GDI-6ch`

rusefillc, 2022-06-30:
yep, naming convention is a mess. but for now we need to look at it :)

NOx-z, 2022-06-30:
Original file no longer exists.. I do see a 6ch folder now existing, but no gerbers generated yet

rusefillc, 2022-06-30:
@NOx-z just edited link

NOx-z, 2022-06-30:
I was expecting HS1/2 to drive LS1/2/3 and HS3/4 to drive LS4/5/6 (or a similar mapping -- edit, it appears HS1/2 is driving LS1/2/5, and HS3/4 is driving LS3/4/6 - and this is okay, but it reads a bit more scatter-brain/using the old mappings and adding onto)

are we making this more variable? like have it selectable as either 4+Pump _OR_ 6 via jumpers/0R, so we have a single board capable of both?

HS5 needs not be in the circuit if no pumping is desired. (I can double check the datasheet, if it wants the pins floating or tied to ground or whatever)
VSENSE3(P/N) needs not be connected/populated, if that microcore is never used for regulation since VSENSE1/2 should take care of the rest. I do not know if microcores can share highside drivers, in which case we want to only drive HS1/2 from the one specific microcore (and thus sense only on one).. but this changes if we want some configurability in there, and I was not aware of that being a goal.. if it is I need to review it as such.

rusefillc, 2022-06-30:
> like have it selectable as either 4+Pump OR 6 via jumpers/0R, so we have a single board capable of both?

yes that's the desire

NOx-z, 2022-06-30:
Okay, let me look it over more with that in mind.. as I did not immediately see where this was done.. immediately I think a big issue will be what is allowed configuration wise and the number 3 sense resistor, and how best to accommodate it.

mck1117, 2022-06-30:
do we think 2 banks is enough to run a 6cyl given the overlapping restrictions?

mck1117, 2022-06-30:
> To me it would make more sense to have INJ5/6 [low sides] tie back to VSENSE1/2 in some jumper manner

Yes - I would much rather see additional transistors with jumpers for both the gate and sense resistor that tie the extra channels in to the same sense resistors as the other injectors as their bank. The current implementation would require much more change to the microcode, as you said.

mck1117, 2022-06-30:
Ok, after some discussion with nozx on slack, here's my proposal.  The current implementation has a problem that firing injector 5 or 6 actually blocks 3 other injectors - since those injectors straddle multiple banks of current sense. Injector 5 blocks injector 6 since it shares a current sense, but it ALSO blocks injectors 1/2 since it shares high side circuitry with them. Likewise, 6 blocks 3/4/5.

These changes also greatly simplify the needed microcode changes.

Proposed changes:
- Source of Q1 connects to R21 instead of R26. This covers injector 5. Injectors 1/2/5 use LS1/2/5 respectively, and all use VSENSE1 for current feedback.
- Add an additional transistor Qx, for injector 6 low side. Selectable jumper between D_LS6 and the drains of Q12 and Qx. In case of pump, install jumper to Q12. In case of injector, install jumper to Qx. Drain goes to R22, shared with channels 3/4. Injectors 3/4/5 use LS3/4/6 respectively, and all use VSENSE2 for current feedback.

[image]

mck1117, 2022-06-30:
With this proposal you can actually run a 5 cylinder with pump (1/2/5 + 3/4), and it only requires moving the single jumper to connect D_LS6 properly.

Qwerty-OFF, 2022-07-03:
@mck1117 can I use single Q12 (without Qx) for PUMP_LOW/INJ6_LOW with this jumpers config?
[image]

mck1117, 2022-07-04:
For the cost of an extra transistor and diode (<$2?), I'd much rather optimize for the minimum possible number of jumpers that you have to flip.  My crappy drawing above only has one jumper to move, while that proposal has 3. Any incorrect combination of jumpers would have unexpected behavior, and could possibly injure the PT2001 and/or transistors.

Qwerty-OFF, 2022-07-04:
I wanted to save some space and make routing easier. And also reduce gate capacitance, if this is critical here. Okay, I'll place an additional transistor if it improves the user experience.