rusefi/rusefi-hardware #109: GDI4 thermal

rusefillc, 2022-06-25:
15A peak 5A hold they say, do we need another round of thermal considerations on the PCBs or are we good?

rusefillc, 2022-07-01:
10 minutes of idle
ambient 24C
board max 36C

```
	// Convert mA to DAC values
	mcUpdateDram(MC33816Mem::Iboost, dacEquation(13000));
	mcUpdateDram(MC33816Mem::Ipeak, dacEquation(9400));
	mcUpdateDram(MC33816Mem::Ihold, dacEquation(3700));
```

I assume 13A boost, 9.4A peak 3.7A hold?

[image]
[image]

mck1117, 2022-07-01:
> I assume 13A boost, 9.4A peak 3.7A hold?

if you're running latest firmware, yes

mck1117, 2022-07-01:
The fact that the PCB is cold over here probably means we need better thermals on the MOSFETs themselves

[image]