rusefi/rusefi-hardware #114: We have GDI4-rev-C fab files

rusefillc, 2022-06-30:
We have GDI4 rev C fab files ready for review

See https://github.com/rusefi/rusefi/wiki/GDI4#changelog for why "rev C"

rusefillc, 2022-07-08:
@mck1117 @NOx-z we have fresh fab files at https://github.com/rusefi/rusefi-hardware/tree/main/GDI-4ch/gerber/rev.%20c including https://github.com/rusefi/rusefi-hardware/blob/main/GDI-4ch/gerber/rev.%20c/GDI-4ch.pdf

NOx-z, 2022-07-08:
The 5Inj+Pump/6Inj schematic seems to read correctly..

a few points that have been discussed but likely lost within chat.

A) I would like to see test points on all (6) the START lines between the asic and the inverter, these can be surface mount (through-hole not needed).

B) Resistor Networks have accumulated solder between leads before, and there was some mild discussion on converting all of these over to standard/non-network resistors.

C) Can we tie the Vcc2P5, pin 20, (actually, the MC33816 is 2.5v, but the PT2001 is 1.5v) to an analog input of the IC? This would be the preferred way to check that the chip has self-powered/regulated instead of a blind wait period that we currently implement in code.

for layout purposes:
D) do not put Q2 in between large capacitors, this is the most common fet to blow and replacement can be difficult

E) please align/group similar test points in straight lines/groups, for instance the SPI test through holes were not aligned in rev A

There are likely more, but these are the ones that come to mind as of right now.

Qwerty-OFF, 2022-07-10:
> Can we tie the Vcc2P5, pin 20, (actually, the MC33816 is 2.5v, but the PT2001 is 1.5v) to an analog input of the IC? This would be the preferred way to check that the chip has self-powered/regulated instead of a blind wait period that we currently implement in code.

Low-side 1 output moved to PB8 to free analog input.

Qwerty-OFF, 2022-07-18:
A) done
B) done
C) done
D) done
E) Done as well as possible without major rework
Ready for fab

rusefillc, 2022-08-05:
fabbed

closed 2022-08-05 by rusefillc