rusefi/rusefi-hardware #125: GDI board: dual digital input

rusefillc, 2022-07-18:
We still have three available pins

Can we please have two channels of protected digital inputs? Probably old friend 74HC2G17GW https://github.com/rusefi/proteus/issues/1

[image]

Qwerty-OFF, 2022-07-21:
Added 4 external channels. Also made jumpers to use the MCU I/O for other purposes.

rusefillc, 2022-07-22:
looking to have at least one DIN as a HW timer, meaning

PA6 MISO totally cannot touch this one
PA8 kind of can touch but let's not touch to keed firmware compatible between revisions
PB6 LS2 let's flip between LS2 and any of the DIN
PA1 currently VCC2P5

please do at least one of the following changes maybe both if small effort

1) PB6 LS2 let's flip between LS2 and any of the DIN
2) VCC2P5 move to any other A* pin and make PA1 a hardware jumper option between A* and DIN, that way we can leave one of current DIN pins as just a via for future use

speaking of which can we please have a via or test pad on some of DIN* and A* to have an option to repurpose those if needed. Just a couple, via would be better, please use your judgement based on level of effort

Qwerty-OFF, 2022-07-22:
[image]

closed 2022-07-22 by Qwerty-OFF