rusefi/rusefi-hardware #128: Gdi4: lower caps are hitting case

rusefillc, 2022-08-04:
Stamped case has this area in the middle which interferes with the lower two caps

Minir hammer mod helps so we need to move just a bit or reduce caps height?

rusefillc, 2022-08-04:
[image]
[image]

Qwerty-OFF, 2022-08-05:
Sad result due to edits https://github.com/rusefi/rusefi-hardware/issues/114
So we need to move capacitors

rusefillc, 2022-08-05:
Would @Qwerty-OFF non-JLC lower caps exist to try in same location? i am fine hand-soldering those two caps

Qwerty-OFF, 2023-05-13:
2x UBY1K471MHL (JLC C5110766 out of stock) - 470uF 80V
2x UPW1V122MHD6TN (JLC C5111952 out of stock) - 1200uF 35V
[image]
@rusefillc please check placement

rusefillc, 2023-05-14:
UPW1V122MHD6TN is pretty available https://www.findchips.com/search/UPW1V122MHD6TN

rusefillc, 2023-05-14:
@Qwerty-OFF placement looks great!