rusefi/rusefi-hardware #130: gdi4-revB: 5v confusion

rusefillc, 2022-08-06:
Q1: what is pin11? schematics says nothing, PCB maybe says 5v?

[image]

Q2: those 5v pins let's clarify on schematics that those are +5v _outputs_ (assuming those are outputs? while playing we can use as inputs but that's not the primary indented use?)

Qwerty-OFF, 2022-08-07:
Q1:
[image]
Q2: how? Like this?
[image]

rusefillc, 2022-08-07:
супер! 

let's do "output". то же самое для пина 11?

Qwerty-OFF, 2023-05-13:
[image]

closed 2023-05-13 by Qwerty-OFF