rusefi/rusefi-hardware #131: DIN1 resistors clarification

rusefillc, 2022-08-06:
with both R113 and R122 populated do we have DIN1 traced to two different stm32 pins by default?

also R113 state "0" while R122 is yellow without a value not, that's a bit confusing

[image]

Qwerty-OFF, 2022-08-07:
In this configuration, DIN1 is connected to two different stm32 pins. To connect A0, just remove R122 and put R121, DIN1 will remain connected to PC13 (which does not have a timer).
R122 is a jumper with soldering holes, so it's yellow. In the future we can use yellow resistors like in hellen boards

rusefillc, 2023-05-14:
Can we add DNP on R122 to make schematics more readable?

Let's add "Use R122 option without R113 for SENT"

Qwerty-OFF, 2023-05-14:
[image]

closed 2023-05-14 by Qwerty-OFF

Qwerty-OFF, 2023-05-14:
R122 now populated
[image]