rusefi/rusefi-hardware #137: SENT: without power we see 10% success rate

rusefillc, 2022-08-08:
ETB connected with two wires: GND and signal wire

ETB +5v is NOT connected ANYWHERE

note a lot of 04 and 12 nibbles what is special about those?

```
[0] 04 04 04 04 04 04 ETB 1220 1092 pos=076 err 028320 001896 s_e=021281 c_err=001174 sync=024508 rate=0087
[0] 12 12 12 04 04 12 ETB 1220 1092 pos=076 err 028507 001904 s_e=021407 c_err=001183 sync=024652 rate=0087
[0] 12 04 04 12 04 04 ETB 1220 1092 pos=076 err 028672 001911 s_e=021518 c_err=001191 sync=024786 rate=0087
[0] 04 04 12 04 04 04 ETB 3148 1220 pos=017 err 028832 001917 s_e=021649 c_err=001200 sync=024911 rate=0087
[0] 04 12 04 04 12 04 ETB 3148 1220 pos=017 err 028988 001928 s_e=021730 c_err=001209 sync=025034 rate=0087
[0] 12 04 04 12 04 12 ETB 3148 1220 pos=017 err 029136 001938 s_e=021863 c_err=001213 sync=025170 rate=0087
[0] 04 12 04 12 04 04 ETB 3148 1220 pos=017 err 029251 001952 s_e=021973 c_err=001220 sync=025306 rate=0087
[0] 12 12 12 12 04 12 ETB 3148 1220 pos=017 err 029358 001964 s_e=022092 c_err=001225 sync=025423 rate=0087
[0] 12 12 04 12 12 04 ETB 3268 1228 pos=013 err 029461 001976 s_e=022174 c_err=001234 sync=025542 rate=0087
[0] 12 12 12 12 04 04 ETB 3268 1228 pos=013 err 029592 001986 s_e=022273 c_err=001243 sync=025652 rate=0087
[0] 04 12 04 04 04 04 ETB 3268 1228 pos=013 err 029700 001997 s_e=022375 c_err=001251 sync=025769 rate=0087
[0] 12 12 04 12 12 04 ETB 3268 1228 pos=013 err 029796 002008 s_e=022458 c_err=001259 sync=025891 rate=0087
[0] 12 12 04 04 12 12 ETB 3268 1228 pos=013 err 029904 002024 s_e=022581 c_err=001263 sync=026009 rate=0087
[0] 12 12 04 12 12 04 ETB 3268 1228 pos=013 err 029998 002039 s_e=022664 c_err=001270 sync=026119 rate=0087
[0] 04 04 04 04 04 04 ETB 3268 1228 pos=013 err 030091 002051 s_e=022755 c_err=001279 sync=026220 rate=0086
[0] 12 12 04 04 04 04 ETB 3268 1228 pos=013 err 030190 002065 s_e=022829 c_err=001285 sync=026326 rate=0086
[0] 12 12 04 12 12 04 ETB 3268 1228 pos=013 err 030292 002081 s_e=022915 c_err=001294 sync=026431 rate=0086

```

rusefillc, 2022-08-08:
@
[image]

rusefillc, 2022-08-08:
apply constant 1.4V from power supply to PA6

actual result: enjoy all nibbles of 04
expected result: no PAL events thus no data

```

[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2831646 089553 s_e=688374 c_err=040238 sync=819172 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2834588 089553 s_e=688405 c_err=040238 sync=819196 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2837580 089553 s_e=688439 c_err=040238 sync=819236 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2840606 089553 s_e=688470 c_err=040238 sync=819274 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2843628 089553 s_e=688512 c_err=040238 sync=819320 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2846585 089553 s_e=688555 c_err=040238 sync=819349 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2849670 089553 s_e=688585 c_err=040238 sync=819376 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2852690 089553 s_e=688618 c_err=040238 sync=819415 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2855746 089553 s_e=688653 c_err=040238 sync=819457 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2858771 089553 s_e=688694 c_err=040238 sync=819502 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2861729 089553 s_e=688736 c_err=040238 sync=819531 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2864718 089553 s_e=688765 c_err=040238 sync=819558 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2867710 089553 s_e=688802 c_err=040238 sync=819600 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2870782 089553 s_e=688835 c_err=040238 sync=819636 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2873810 089553 s_e=688875 c_err=040238 sync=819684 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2876860 089553 s_e=688919 c_err=040238 sync=819716 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2879847 089553 s_e=688952 c_err=040238 sync=819743 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2882812 089553 s_e=688986 c_err=040238 sync=819785 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2885880 089553 s_e=689021 c_err=040238 sync=819822 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2888945 089553 s_e=689064 c_err=040238 sync=819872 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2891952 089553 s_e=689104 c_err=040238 sync=819900 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2894984 089553 s_e=689136 c_err=040238 sync=819928 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2897996 089553 s_e=689169 c_err=040238 sync=819967 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2901080 089553 s_e=689201 c_err=040238 sync=820002 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2904130 089553 s_e=689241 c_err=040238 sync=820049 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2907063 089553 s_e=689284 c_err=040238 sync=820081 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2910043 089553 s_e=689316 c_err=040238 sync=820108 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2913013 089553 s_e=689346 c_err=040238 sync=820144 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2916088 089553 s_e=689378 c_err=040238 sync=820183 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2919168 089553 s_e=689422 c_err=040238 sync=820234 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2922152 089553 s_e=689467 c_err=040238 sync=820262 rate=0084
[0] 04 01 04 12 04 04 ETB 1220 1092 pos=076 err 2925126 089553 s_e=689500 c_err=040238 sync=820295 rate=0084
```

rusefillc, 2022-08-08:
https://electronics.stackexchange.com/questions/394846/stm32-logic-level-high-and-low-thresholds

```
The logic levels are described in section 6.3.14. When your supply voltage is 3.3V then:

V_IL = 0.39 * Vcc - 0.06 = 1.23 V (table row "I/O input low level voltage except BOOT0". The maximum voltage that will be read as logical zero is 1.23V (+ read all footnotes).

V_IH = 0.49 * Vcc + 0.26 = 1.88 V (table row "I/O input high level voltage except BOOT0"). You need to supply at lest 1.88 V to make the pin read as logical one.

Your 1.8 V signal is below V_IH and above V_IL, which means that the result is unpredictable.
```

so on the one had 1.4 is unpredictable

problem: I get unhappy results with 0.5v as well

rusefillc, 2022-08-08:
@dron0gus very tempting to ask for a bit of help :(

https://github.com/rusefi/rusefi-hardware/tree/main/SENT-box/firmware
master
blue pil

SENT input pin first channel PA6 as defined at https://github.com/rusefi/rusefi-hardware/blob/main/SENT-box/firmware/io/io_pins.h

UART connectivity on the bottom of same https://github.com/rusefi/rusefi-hardware/blob/main/SENT-box/firmware/io/io_pins.h

default FW uses PAL input

majority of code at https://github.com/rusefi/rusefi-hardware/blob/main/SENT-box/firmware/sent.cpp with UART logging at https://github.com/rusefi/rusefi-hardware/blob/main/SENT-box/firmware/uart.cpp

dron0gus, 2022-08-10:
FIY: PA6 and PA7 are not 5V tolerant pins. Pulling it to 5V may damage device.
PB4 and PB5 have same TIM3 functions. but 5V tolerant.