rusefi/rusefi-hardware #143: GDI4 rev C quality issues

rusefillc, 2022-08-14:
At least two boards got damaged on first injector bench test

@NOx-z reports one board not driving injectors 3 and 4 probably due to poor soldering of D11

[image]
[image]

i just had a board make a pop sound while testing injector 4 :( on that board injectors 1, 2 and 3 were working fine. I was using rusEFI Proteus to produce "proper" short test signals and https://github.com/rusefi/rusefi-hardware/releases/tag/20220604_release firmware

rusefillc, 2022-08-14:
@Qwerty-OFF any ideas what can we make to increase our fabrication chances? would wider pads cause more solder paste? do we have control over solder paste amount?

rusefillc, 2022-08-14:
Just found a second board with RESET button not working. Looks like JLC has messed up this batch :(

rusefillc, 2022-08-14:
@mck1117 do "reset" and "boot" buttons work on your board? two of mine needed extra solder for RESET to start working

@NOx-z  do "reset" and "boot" buttons work on your board? two of mine needed extra solder for RESET to start working

NOx-z, 2022-08-14:
I just poked at it, and my BOOT button completely fell off, then I poked the reset, and it lifted off one leg..

rusefillc, 2022-08-14:
wow :(

something about solder paste went very wrong on this order. just filed a quality complaint.

rusefillc, 2022-08-15:
[image]
[image]

rusefillc, 2022-08-16:
[Noxz](https://app.slack.com/team/UBFRN2SH0)
  [2:52 PM](https://rusefi.slack.com/archives/CFP3QAJCW/p1660589556211969)
I dont really have experience with production settings to answer this next bit.. it seems more an issue with quality control than board layout or components (unless, for instance, the plating/metal of the buttons did not want to adhere to the solder well)

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)
  [2:56 PM](https://rusefi.slack.com/archives/CFP3QAJCW/p1660589771313479)
I think it's a fools errand to try and fully self-test power electronics

[Noxz](https://app.slack.com/team/UBFRN2SH0)
  [3:03 PM](https://rusefi.slack.com/archives/CFP3QAJCW/p1660590191173269)
beyond making a jig with pogo pins all over to check continuity autoamtically..

[Matthew Kennedy](https://app.slack.com/team/U80K9BYVD)
  [3:03 PM](https://rusefi.slack.com/archives/CFP3QAJCW/p1660590226097949)
Ever look at a real ECU? Yeah, that's what they do. Test pads all over the place.
[3:03](https://rusefi.slack.com/archives/CFP3QAJCW/p1660590238188559)
but that's what I mean by self test
[3:04](https://rusefi.slack.com/archives/CFP3QAJCW/p1660590249185179)
some things you can self test by adding some analog switches to feed your output back in to the input
[3:04](https://rusefi.slack.com/archives/CFP3QAJCW/p1660590260797269)
but that only works when the penalty for "bad board" is not working, instead of releasing vast magic smoke

mck1117, 2022-08-17:
Backwards diode on Andrey's rev C

[image]

mck1117, 2022-08-17:
Correct board from the same batch, for reference

[image]

rusefillc, 2022-08-17:
Wow, that's a third kind of issue on same batch :(

mck1117, 2022-08-17:
can you forward the images from this thread on to JLC to try and get our coin back on this batch?

closed 2023-05-14 by rusefillc