rusefi/rusefi-hardware #146: GM fuel pressure decoding

rusefillc, 2022-08-15:
ambient room

```

```

10 bar = 1000 KPa

```
nibbles: ST=0 D=1:1:A:B:1:1 CRC=8 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 122949
 GM: St 4, Sig0 0282, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=4 D=1:1:9:B:1:1 CRC=4 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 123158
 GM: St 0, Sig0 0281, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EB (747), T = 23.34375C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:A:C:1:1 CRC=0 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 123367
 GM: St 8, Sig0 0282, Sig1 3089. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:A:C:1:1 CRC=0 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 123579
 GM: St C, Sig0 0282, Sig1 3089. Tick = 2541 nS.
  msg 16: 0x02EB (747), T = 23.34375C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:A:C:1:1 CRC=0 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 123791
 GM: St 4, Sig0 0282, Sig1 3089. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:A:B:1:1 CRC=8 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 124002
 GM: St C, Sig0 0282, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EB (747), T = 23.34375C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=0 D=1:1:A:C:1:1 CRC=8 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 124211
 GM: St 4, Sig0 0282, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=4 D=1:1:9:B:1:1 CRC=4 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 124421
 GM: St 4, Sig0 0281, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:9:B:1:1 CRC=4 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 124630
 GM: St 8, Sig0 0281, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:A:B:1:1 CRC=8 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 124839
 GM: St C, Sig0 0282, Sig1 2833. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=4 D=1:1:B:C:1:1 CRC=0 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 125050
 GM: St 0, Sig0 0282, Sig1 3089. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=C D=1:1:A:D:1:1 CRC=7 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 125260
 GM: St C, Sig0 0282, Sig1 3345. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=4 D=1:1:A:C:1:1 CRC=0 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 125471
 GM: St 4, Sig0 0282, Sig1 3089. Tick = 2541 nS.
  msg 16: 0x02EA (746), T = 23.31250C ?
  msg 22: 0x02ED (749), T = 23.40625C ?
nibbles: ST=0 D=1:1:B:C:1:1 CRC=0 Errs Short: 000029 Long: 000003 Sync: 000001 CRC: 000001 (000 %). Frames: 125679
```

rusefillc, 2022-08-15:
[image]

rusefillc, 2022-08-15:
10 bar looks great
```
nibbles: ST=8 D=1:1:E:F:1:1 CRC=4 Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 158447
 GM: St 8, Sig0 0286, Sig1 0287, 9.650 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=C D=1:1:E:F:1:1 CRC=4 Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 158653
 GM: St 4, Sig0 0286, Sig1 0287, 9.650 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=8 D=1:1:E:0:2:1 CRC=8 Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 158863
 GM: St 8, Sig0 0286, Sig1 0288, 9.700 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=0 D=1:1:F:0:2:1 CRC=C Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 159071
 GM: St 0, Sig0 0287, Sig1 0288, 9.750 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=0 D=1:1:E:0:2:1 CRC=4 Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 159281
 GM: St 4, Sig0 0286, Sig1 0287, 9.650 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=C D=1:1:E:0:2:1 CRC=8 Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 159491
 GM: St 8, Sig0 0286, Sig1 0288, 9.700 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=C D=1:1:E:0:2:1 CRC=8 Errs Short: 000008 Long: 000004 Sync: 000000 CRC: 000000 (000 %). Frames: 159702
 GM: St C, Sig0 0286, Sig1 0288, 9.700 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=0 D=1:1:E:0:2:1 CRC=8 Errs Short: 000008 Long: 000005 Sync: 000000 CRC: 000000 (000 %). Frames: 159909
 GM: St 4, Sig0 0286, Sig1 0288, 9.700 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=8 D=1:1:E:0:2:1 CRC=8 Errs Short: 000008 Long: 000005 Sync: 000000 CRC: 000000 (000 %). Frames: 160119
 GM: St 8, Sig0 0286, Sig1 0288, 9.700 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=0 D=1:1:F:0:2:1 CRC=C Errs Short: 000008 Long: 000005 Sync: 000000 CRC: 000000 (000 %). Frames: 160329
 GM: St 0, Sig0 0287, Sig1 0288, 9.750 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
nibbles: ST=0 D=1:1:E:0:2:1 CRC=8 Errs Short: 000008 Long: 000005 Sync: 000000 CRC: 000000 (000 %). Frames: 160537
 GM: St 0, Sig0 0286, Sig1 0288, 9.700 Atm Tick = 2541 nS.
  msg 16: 0x02F3 (755), T = 23.59375C ?
  msg 22: 0x02F6 (758), T = 23.68750C ?
```

no pressure - probably should it be close to 1 @dron0gus ? what does it show on your end?
```
nibbles: ST=0 D=0:C:0:2:C:0 CRC=8 Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 062893
 GM: St 0, Sig0 0192, Sig1 0194, 0.300 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:1:2:C:0 CRC=C Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 063093
 GM: St 4, Sig0 0193, Sig1 0194, 0.350 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=C D=0:C:1:2:C:0 CRC=C Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 063294
 GM: St C, Sig0 0193, Sig1 0194, 0.350 Atm Tick = 2541 nS.
  msg 16: 0x02F1 (753), T = 23.53125C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:1:2:C:0 CRC=C Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 063495
 GM: St 0, Sig0 0193, Sig1 0194, 0.350 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:1:2:C:0 CRC=B Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 063695
 GM: St 0, Sig0 0193, Sig1 0195, 0.400 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=8 D=0:C:0:3:C:0 CRC=F Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 063896
 GM: St C, Sig0 0192, Sig1 0195, 0.350 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=8 D=0:C:1:2:C:0 CRC=8 Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 064096
 GM: St C, Sig0 0192, Sig1 0194, 0.300 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:0:2:C:0 CRC=8 Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 064297
 GM: St 0, Sig0 0192, Sig1 0194, 0.300 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:1:2:C:0 CRC=C Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 064497
 GM: St 4, Sig0 0193, Sig1 0194, 0.350 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=C D=0:C:1:3:C:0 CRC=B Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 064698
 GM: St C, Sig0 0193, Sig1 0195, 0.400 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:0:2:C:0 CRC=8 Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 064898
 GM: St 4, Sig0 0192, Sig1 0194, 0.300 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?
nibbles: ST=0 D=0:C:1:2:C:0 CRC=C Errs Short: 000008 Long: 000002 Sync: 000000 CRC: 000000 (000 %). Frames: 065099
 GM: St 0, Sig0 0193, Sig1 0194, 0.350 Atm Tick = 2541 nS.
  msg 16: 0x02F2 (754), T = 23.56250C ?
  msg 22: 0x02F4 (756), T = 23.62500C ?

```

dron0gus, 2022-08-16:
No pressure. Saint-P. Almost sea level.
Bouncing around 1 Atm.
Seems sensor is not very precise at low pressure?
```
nibbles: ST=0 D=0:C:5:B:C:0 CRC=8 Errs Short: 000026 Long: 000000 Sync: 000000 CRC: 000000 (000 %). Frames: 564959
 GM: St 4, Sig0 0198, Sig1 0203, 1.050 Atm Tick = 2597 nS.
  msg 16: 0x0303 (771), T = 24.09375C ?
  msg 22: 0x0302 (770), T = 24.06250C ?
nibbles: ST=4 D=0:C:5:B:C:0 CRC=4 Errs Short: 000026 Long: 000000 Sync: 000000 CRC: 000000 (000 %). Frames: 565150
 GM: St 4, Sig0 0197, Sig1 0203, 1.000 Atm Tick = 2597 nS.
  msg 16: 0x0303 (771), T = 24.09375C ?
  msg 22: 0x0302 (770), T = 24.06250C ?
nibbles: ST=C D=0:C:5:A:C:0 CRC=3 Errs Short: 000026 Long: 000000 Sync: 000000 CRC: 000000 (000 %). Frames: 565340
 GM: St C, Sig0 0197, Sig1 0202, 0.950 Atm Tick = 2597 nS.
  msg 16: 0x0303 (771), T = 24.09375C ?
  msg 22: 0x0302 (770), T = 24.06250C ?
nibbles: ST=0 D=0:C:5:B:C:0 CRC=4 Errs Short: 000026 Long: 000000 Sync: 000000 CRC: 000000 (000 %). Frames: 565530
 GM: St 8, Sig0 0197, Sig1 0203, 1.000 Atm Tick = 2597 nS.
  msg 16: 0x0303 (771), T = 24.09375C ?
  msg 22: 0x0302 (770), T = 24.06250C ?
nibbles: ST=C D=0:C:6:B:C:0 CRC=8 Errs Short: 000026 Long: 000000 Sync: 000000 CRC: 000000 (000 %). Frames: 565721
 GM: St C, Sig0 0198, Sig1 0203, 1.050 Atm Tick = 2597 nS.
  msg 16: 0x0303 (771), T = 24.09375C ?
  msg 22: 0x0302 (770), T = 24.06250C ?
```

mck1117, 2022-08-16:
>  Seems sensor is not very precise at low pressure?

How high does this sensor go? If it's a GDI sensor, I'd expect at least 200 bar, so 0.05 bar resolution isn't terrible. 200/0.05 is 12 bits.

dron0gus, 2022-08-16:
Actual resolution seems 1/10 Atm
See https://github.com/rusefi/rusefi-hardware/blob/fc7550c533f5a8c9343e8c53ae82b3afd78bb808/SENT-box/firmware/sent.cpp#L458
But we can get 0.05 by averaging two signals.

floods57, 2024-05-07:
Hi, 
Can you share the code you use in the Blue Pill to test ? i looking to make a similar test on a Audi and a Renault intake pressure/temp sensor that use SENT protocol too with 12 bits data (12 bits for pressure and 12 bits for temp).
Thanks

rusefillc, 2024-05-07:
@floods57 all the code is at https://github.com/rusefi/rusefi-hardware/tree/main/SENT-box/firmware

dron0gus, 2024-05-08:
@floods57 can you share partnumber(s) of these sensors?

floods57, 2024-05-08:
Thanks a lot ;)
About the partnumbers yes, the Renault sensor is a Sensata 82MPP12-01, the Audi, i have 2, the 1st is a Sensata 04E906051J (82MPP07-03) and the second is a Valeo 04E906051H, the 2 sensor are on the 1.5 TSI, one is the MAP, and the second is before throttle body (boost sensor), i don't know why they use 2 different brand, but many chance there give the exact same data.

dron0gus, 2024-05-08:
@floods57 thanks for partnumbers.
Second one from Audi is 10 times cheaper then first. At least for used one. Bought two for price of 3 cups of coffee.
If you know other sensors with SENT - please let me know.
Also please note that most recent driver is in main RusEFI codebase: https://github.com/rusefi/rusefi/tree/master/firmware/hw_layer/drivers/sent

floods57, 2024-05-08:
Ok, but i am not sure about the code i must use, i looking for the same as in this video : [https://www.youtube.com/watch?v=S2xH2oOEYL8](https://www.youtube.com/watch?v=S2xH2oOEYL8) i looking to do a similar testing with a Blue Pill to 1st find the value of the sensor.
Ok about other sensor, i think i can find more, i know there is a other reference for the Renault 1.3 liter, must check, will let you know, at this time i have this 3 in stock for testing ;)
Thanks

dron0gus, 2024-05-13:
@floods57 for bluepill use https://github.com/rusefi/rusefi-hardware/tree/main/SENT-box/firmware code. But check main RusEFI FW for possible updates, changes.
BTW, do you have pinout for these map/iat sensor? Just received one for testing.

dron0gus, 2024-05-13:
04E906051H
Room temperature (a bit hot), sea level atm preasure.
Seems Sig0 is pressure, Sig1 is temperature.
Slow channels does not change (partnumber? serial number?) Can we use it for sensor identification?
```
2024-05-14_00_50_24_401: EngineState: ---- SENT ch 0 ---- on 23 - AN temp 2
2024-05-14_00_50_24_402: EngineState: Unit time 486 CPU ticks 2.892857088 uS
2024-05-14_00_50_24_402: EngineState: Total pulses 2982654
2024-05-14_00_50_24_403: EngineState: Last valid fast msg Status 0x4 Sig0 0x41A Sig1 0x312
2024-05-14_00_50_24_404: EngineState: Slow channels:
2024-05-14_00_50_24_404: EngineState:  ID 35: 0
2024-05-14_00_50_24_405: EngineState:  ID 41: 2093
2024-05-14_00_50_24_406: EngineState:  ID 148: 2577
2024-05-14_00_50_24_409: EngineState:  ID 151: 1262
2024-05-14_00_50_24_410: EngineState:  ID 4: 1
2024-05-14_00_50_24_410: EngineState:  ID 7: 83
2024-05-14_00_50_24_413: EngineState:  ID 8: 324
2024-05-14_00_50_24_413: EngineState:  ID 129: 0
2024-05-14_00_50_24_414: EngineState:  ID 10: 3896
2024-05-14_00_50_24_414: EngineState:  ID 1: 0
2024-05-14_00_50_24_417: EngineState:  ID 147: 1360
2024-05-14_00_50_24_417: EngineState:  ID 150: 3281
2024-05-14_00_50_24_417: EngineState:  ID 9: 193
2024-05-14_00_50_24_420: EngineState:  ID 144: 1296
2024-05-14_00_50_24_420: EngineState:  ID 146: 1424
2024-05-14_00_50_24_421: EngineState:  ID 3: 7
2024-05-14_00_50_24_423: EngineState: Restarts 0
2024-05-14_00_50_24_423: EngineState: Interval errors 136828 short, 5133 long
2024-05-14_00_50_24_425: EngineState: Total frames 277008 with CRC error 6100 (2.202102416%)
2024-05-14_00_50_24_426: EngineState: Total slow channel messages 3787 with crc6 errors 26 (0.686559232%)
2024-05-14_00_50_24_426: EngineState: Sync errors 7215
2024-05-14_00_50_24_428: EngineState: --------------------
```

dron0gus, 2024-05-13:
Pinout:
1 - SENT
2 - ground
3 - power (5V)

floods57, 2024-05-15:
Ok, Thanks i will try it ;)
For the pinout, i have it for the Sensata 82MPP12-01 (pin 1 is signal output, 2 is GND and 3 is +5V)

floods57, 2024-05-15:
> Pinout: 1 - SENT 2 - ground 3 - power (5V)

Ok, so many chance that the 04E906051J use the same pinout, will try it, look really difficult to get any information for this sensor.
As i know from the 82MPP12-01, the slow channel is not used, and 1st data (3 nibbles after the status) is the pressure and next the temp, and i just don't understand how the status work on this.

floods57, 2024-05-16:
@dron0gus By the way, did you use a RC filter ? Thanks

dron0gus, 2024-05-16:
@floods57 do you mean input RC filter? No, I'm debuging on MRE board and using analog input with filter C removed. So signal path is OpAmp with K = 1 -> 1/2 divider -> STM32 pin.
Actually maybe because of this I see about 2% of error frames.

rusefillc, 2024-10-23:
Just started to extract code towards https://github.com/rusefi/rusefi/issues/6997

closed 2024-10-23 by rusefillc