rusefi/rusefi-hardware #147: GDI4revC WA1 board damange

rusefillc, 2022-08-15:
@NOx-z what components were damangers on your board due to above? I wonder if we can make schematics or code smarter to survivei?

[image]

rusefillc, 2022-08-15:
@mck1117 @Qwerty-OFF FYI

rusefillc, 2022-08-15:
related to #143

NOx-z, 2022-08-15:
the PT2001 fried twice before I found the issue, but the first time also blew out Q2 (HV fet) and it's sense resistor R12 - this occurred after a long on-time (with DrvEn pulled low/disabled) with a busted asic. The second time only blew out the asic before I removed the power, swapped it out, and everything else was okay.

rusefillc, 2022-08-15:
TL,DR: too much drama with that board, no clear data?

NOx-z, 2022-08-15:
We've been able to click all the injectors. We have at least 2 fully working boards (well, once I slap the buttons back on). The new firmware thus far is also operating as expected. We resolved many issues from Rev B, but even that had fab issues

rusefillc, 2022-08-15:
@NOx-z my main question here: can we think of mods to make boards more resilient against fab issues?