rusefi/rusefi-hardware #151: single WBO rev 3 PCB is a little too large

rusefillc, 2022-08-23:
rev 3 is 
57.8 wide
66.4 mm long

width has to be belo
54.2 mm

please cut 1.8mm on both sides of the PCB thus shrink PCB width by 3.6mm

looks like length is OK but let's trim 0.5 just to be a bit more sure

rusefillc, 2022-08-30:
@Nadezhda-n also at least four capacitors were added under #153 and some stm32 pins were moved around, looking for update PCB to match current schematics :)

Nadezhda-n, 2022-08-31:
done :)

rusefillc, 2022-08-31:
@dron0gus please review in term of capacitors

@Nadezhda-n please confirm that you had a chance to try your gerbers. Please generate gerbers/bom/CPL and push into "rev 4" folder"

Nadezhda-n, 2022-08-31:
added folder export/rev4

rusefillc, 2022-08-31:
@Nadezhda-n please change 

U11,U12 part number to C840103