rusefi/rusefi-hardware #157: digital inputs board: CAN TX

rusefillc, 2022-09-30:
https://github.com/rusefi/rusefi-hardware/tree/main/digital-inputs/firmware firmware should be sending (and later transmitting) CAN stuff at 500Kb in order to control hardware under test

rusefillc, 2022-09-30:
@dron0gus I've pushed something but logic analyzer does not show happy CAN outbound traffic. Is it CAN registers, pin init or something else?

https://github.com/rusefi/rusefi-hardware/commit/160935062495c95033a3ee4b34da6a16801284e3

rusefillc, 2022-09-30:
[image]

rusefillc, 2022-09-30:
[image]

dron0gus, 2022-10-02:
CAN needs xmitter to be attached - it listens to what is transmits. I see no xmitter on photo.
CAN needs at least two nodes - other end should ack valid messages. Without ack HW will go to error state or infinity transmit loop.
CAN needs load resistor.