rusefi/rusefi-hardware #159: quad DDPAK igbt

rusefillc, 2022-11-29:
https://github.com/rusefi/rusefi/tree/master/hardware/DDPAK_breakout

rusefillc, 2022-11-29:
See also https://github.com/rusefi/hellen154hyundai/ 

[image]

rusefillc, 2022-11-29:
@Nadezhda-n please PR into https://github.com/rusefi/rusefi-hardware/tree/main/quad-igbt folder :)

Nadezhda-n, 2022-12-02:
[image]
[image]

rusefillc, 2022-12-02:
@Nadezhda-n у GND будет дырочка P9?

rusefillc, 2022-12-02:
@Nadezhda-n R2 в оригинале между R1 и IGBT

а стало 100K снаружи 1K

кажется удобнее и более гибко будет R2 между R1 и IGBT

Nadezhda-n, 2022-12-02:
[image]

Nadezhda-n, 2022-12-02:
> @Nadezhda-n please PR into https://github.com/rusefi/rusefi-hardware/tree/main/quad-igbt folder :)

Посмотри, верно ли тебя поняла :)

rusefillc, 2022-12-02:
@Nadezhda-n it looks great!

Please add "rusefi.com" silkscreen on the back and push JLC fab files

@Qwerty-OFF please review

rusefillc, 2022-12-05:
@mck1117 @dron0gus does this need capacitors like https://github.com/andreika-git/hellen-one/issues/207#issuecomment-1336519453 ?

dron0gus, 2022-12-05:
@rusefillc this capacitor is in feedback circuit, not base

rusefillc, 2023-11-16:
we have the boards

closed 2023-11-16 by rusefillc