rusefi/rusefi-hardware #160: Улучшить 154 плату

rusefillc, 2022-12-04:
https://github.com/rusefi/rusefi-hardware/tree/main/breakout-boards/Breakout_154pin_284617-1-Connector

1) нужно увеличить диаметр маленьких отверстий со нижней половины платы (там, где провода) с 1мм до 1.2мм - не уменьшая ширину дорожек. Для этого видимо ряды 51-60 и 133-154 придётся поставить в шахматном порядке?

2) на шёлкографии нужно подписать сильно больше пинов - в идеале найти способ подписать все пины проводов с обеих сторон и все пины разъёма сзади

3) версию платы на шёлкографии поменять с 0.3 на 0.4

Nadezhda-n, 2022-12-08:
[image]
не очень поняла откуда такие номера в п.1
Если опечатка в числах, то да, надо ставить в шахматном

Nadezhda-n, 2022-12-08:
[image]

rusefillc, 2022-12-13:
ordered

closed 2022-12-13 by rusefillc