rusefi/rusefi-hardware #161: WBO rev 3: heater PWM is oscillating in closed loop

dron0gus, 2022-12-04:
Measured Rint and sensor temperature are too noisy for default PID settings.
This cause heater PWM to oscillate from 30% to 90% with 0.5..1 Hz period.
This cause waving consumed current.
TODO: debug HW.
SW fix:
```
diff --git a/firmware/heater_control.cpp b/firmware/heater_control.cpp
index b97d923..628f066 100644
--- a/firmware/heater_control.cpp
+++ b/firmware/heater_control.cpp
@@ -48,8 +48,8 @@ static struct heater_state state[AFR_CHANNELS] =
 {
     {
         .heaterPid = Pid(
-            0.3f,      // kP
-            0.3f,      // kI
+            0.03f,      // kP
+            0.03f,      // kI
             0.01f,     // kD
             3.0f,      // Integrator clamp (volts)
             HEATER_CONTROL_PERIOD

```

rusefillc, 2022-12-04:
@mck1117 fyi

mck1117, 2022-12-04:
Can you show a log of this happening, either from rusEFI or wb controller directly?

mck1117, 2022-12-07:
@rusefillc wideband controller's own TS config now logs various controller parameters, give it a shot?

[image]

rusefillc, 2022-12-07:
My soldered piece is in the suit case as a talking piece at PRI, I might solder another one tomorrow. @dron0gus any chance for logs?

rusefillc, 2022-12-07:
@mck1117 realistically I need to waste time on different stuff, I do need help with this WBO :( Can I order larger stm32 chips for you or did you have a chance?

mck1117, 2022-12-07:
FWIW my dual board works fine, I don't remember if anything is substantially different on rev3

I can also attempt to flash my gd32 rev3 and see what happens

rusefillc, 2022-12-07:
@dron0gus wonder is tracing noise is the cause

mck1117, 2022-12-07:
> @dron0gus wonders is tracing noise is the cause

This seems sort of unlikely, none of the nodes are particularly low impedance.

dron0gus, 2022-12-07:
@mck1117 thanks for fixing logging.
I'll try to take logs this evening.
Happens only on f1_rev3.
Dual works fine with default PID settings.

BTW should we export PIDs settings to TS so user will be able to adjust.

mck1117, 2022-12-07:
I'd love a log with data rate maxed out to confirm whether it's really too noisy or not

dron0gus, 2022-12-07:
@mck1117 
[image]

[2022-12-08_00.12.44.zip](https://github.com/rusefi/rusefi-hardware/files/10180082/2022-12-08_00.12.44.zip)

mck1117, 2022-12-07:
The noise is on your battery voltage measurement, not on the ESR measurement

mck1117, 2022-12-07:
@dron0gus for a test, can you replace [this line](https://github.com/mck1117/wideband/blob/1e3a227f5038e8c418843d0ed7d4c9392de0284e/firmware/heater_control.cpp#L215) with `float batteryVoltage = 12;`? That will disable battery voltage compensation to rule out battery measurement issues.

dron0gus, 2022-12-13:
[image]
This noise may be related.

mck1117, 2022-12-14:
>  This noise may be related.

what node is that?

dron0gus, 2022-12-14:
> what node is that?
Un_3x_sense
Look like 350KHz bandwidth of GS8331/2 is not good for STM32 ADC:
[image]
Also
[image]