rusefi/rusefi-hardware #162: WBO rev3: U11, U12 wrong component mounted

dron0gus, 2022-12-12:
I see something with 5 pins only on my board.

dron0gus, 2022-12-12:
[image]

rusefillc, 2022-12-13:
```

74HC2G17GW | U11 | Package_TO_SOT_SMD:SOT-23-6 | C840103

74HC2G17GW | U12 | Package_TO_SOT_SMD:SOT-23-6 | C840103

```

replacement part SN74LVC1G06DBVR is in fact SOT-23-5 while 74HC2G17GW is TSSOP6

dron0gus, 2022-12-13:
Replacement is "Single Inverter Buffer/Driver With Open-Drain" while original part is "Dual non-inverting Schmitt trigger".
So dual vs single.
Non-inverting vs inverting - ok, SW can take care.
Push-pull output vs open-drain.

dron0gus, 2022-12-13:
[image]
Pinout is not compatible.