rusefi/rusefi-hardware #170: WBO x2 BT documentation

rusefillc, 2023-01-23:
@dron0gus how does one initialize BT on WBOx2?

Looking to add into into https://github.com/rusefi/rusefi-hardware/tree/main/lambda-x2 readme.md around https://github.com/rusefi/rusefi-hardware/tree/main/lambda-x2#bluetooth

Also what is default baud rate?

dron0gus, 2023-01-23:
https://github.com/dron0gus/wideband/blob/master/firmware/boards/f1_dual/wideband_board_config.h#L71 and https://github.com/dron0gus/wideband/blob/2318bd1e01328d1e752bdb59344bb1f8bbec9a0b/firmware/console/binary/tunerstudio_io_serial.cpp#L69 FW will attempt to setup BT module on each power up.

mck1117, 2023-01-23:
> FW will attempt to setup BT module on each power up.

@dron0gus PR?

dron0gus, 2023-01-23:
@mck1117 already in your repo https://github.com/mck1117/wideband/blob/master/firmware/console/binary/tunerstudio_io_serial.cpp .

rusefillc, 2023-01-23:
@dron0gus just improved https://github.com/rusefi/rusefi-hardware/tree/main/lambda-x2#bluetooth a bit

closed 2023-01-23 by rusefillc