rusefi/rusefi-hardware #179: 20A DC motor control

rusefillc, 2023-02-19:
Tommy says "fancy aftermarket wastegates need 20A H-bridges"

@blundar says "20A? That you can actually buy? Nope. Predriver and discrete mosfets"

rusefillc, 2023-04-02:
https://830533.app.netsuite.com/core/media/media.nl?id=15177070&c=830533&h=EZmq3ZpXEflfYctiaH8D5B-ob6L4zEoKnJaOmCgiibSyfkGA&_xt=.pdf

[Data Sheet GenV PowerGate60 eWG.pdf](https://github.com/rusefi/rusefi-hardware/files/11133639/Data.Sheet.GenV.PowerGate60.eWG.pdf)

rusefillc, 2023-04-03:
[image]

rusefillc, 2023-04-06:
@Qwerty-OFF suggest we go (dual) IR2302

https://www.infineon.com/dgdl/Infineon-IR2302-DataSheet-v01_00-EN.pdf?fileId=5546d462533600a4015355c988b216de

rusefillc, 2023-04-10:
@Qwerty-OFF I have Infineon Technologies IR2302STRPBF JLCPCB Part #: C148147 coming

Qwerty-OFF, 2023-04-28:
ETB:
- ШИМ на SD1
- MOT1 управляется IN1, "0" - 0В; "1" - 12В
- MOT2 аналогично с IN2.
Чтобы крутить мотором IN1 и IN2 должны иметь противоположные значения.

HS/LS:
- IN определяет HS или LS будет ("0" - LS; "1" - HS)
- "1" на SD собственно активирует ключ
- Выход это OUT

Питание подается на GND и +12, не забудь подключить провод GND от Nucleo или что там будет в качестве тестовой платы.