rusefi/rusefi-hardware #206: тестовая плата VR или Hall

rusefillc, 2023-05-21:
Пока мы мучаемся https://github.com/andreika-git/hellen-one/issues/285 нужно сделать вариант попроще, понадёжнее и побыстрее

вместо одного универсального входного пина я согласен на три пина - 2*vr+1hall, с единым выходом на процессор

можем ли мы выход VR и выход hall просто XOR микросхемой 74 серии? в реальности будут либо VR подсоединены, либо hall - никогда не оба

* главный разъём 24 пина как на https://github.com/rusefi/alphax-2chan 
* max9926 - два простых канала в стиле https://github.com/rusefi/rusefi-hardware/tree/main/classic-designs/frankenso
* два hall входа

* 1.5мм via для пайки 5v GND outputs проводов

Qwerty-OFF, 2023-05-24:
Распиновка 24 пина любая?
Светодиоды для индикации состояний линий поставить?

rusefillc, 2023-05-24:
Любая

Ставить

rusefillc, 2023-06-01:
всё-таки MAX9924