rusefi/rusefi-hardware #208: GD4-D does not click pump

rusefillc, 2023-06-10:
i have three boards assembled
all three boards click inj1 when I ground inj1 control line

all boards do not click same injector connected 82/83 and 84/85 when I ground pump control pin 41

rusefillc, 2023-06-10:
@Qwerty-OFF @NOx-z @mck1117 if I hook signal generator to pin 41 (proteus with self-stimulation) what easily accessible points can I probe on the board with what expected behavior?

Qwerty-OFF, 2023-06-10:
[image]
[image]
[image]

rusefillc, 2023-06-10:
signal reaches point 3
[image]

rusefillc, 2023-06-10:
q11 and r19 silence high level

[image]

rusefillc, 2023-06-10:
q12 r20 silence low level
[image]

rusefillc, 2023-06-10:
@NOx-z says it's a software issue it wants start6... not start5
I've tested rev A and got pump to click via R60 (start6 input)

rusefillc, 2023-06-10:
workaround: rev D clicks pump via injector 6 control line
proper fix later: microcode change

no HW changes planned for now

rusefillc, 2024-10-20:
Change of plan: let's forever stay as is with pump being controlled by START6.

closed 2024-10-20 by rusefillc