rusefi/rusefi-hardware #219: failing to export gerbers, failing to fail after failing to export gerbers

rusefillc, 2023-07-15:
I've created https://github.com/rusefi/rusefi-hardware/blob/main/.github/workflows/create-board.yaml just to export https://github.com/rusefi/rusefi-hardware/tree/main/breakout-boards/Breakout_154pin_284617-1-Connector for now

* it fails to export see error messages at https://github.com/rusefi/rusefi-hardware/actions/runs/5559778665/jobs/10156227662
* job is green even while it has failed

rusefillc, 2023-07-15:
@chuckwagoncomputing https://github.com/andreika-git/hellen-one/commit/447c4a6fb540b37780b0acbe156d1dfdeb3d21fb and https://github.com/andreika-git/hellen-one/commit/dac43765ba542b6783931610ce00454730465f52 but I am getting closer to the limit of my abilities here :)

rusefillc, 2023-07-15:
@chuckwagoncomputing I think I've fixed the "fails but does not fail job" part. What am I doing wrong with actual export here?...

rusefillc, 2023-07-15:
@chuckwagoncomputing I've made some progress looks like VRML export is failing?

chuckwagoncomputing, 2023-07-15:
It's showing an error, but the ExportVRML_File function is reporting success, and it's outputting a VRML file. Is the file not correct?

chuckwagoncomputing, 2023-07-15:
When I view the vrml file with https://imagetostl.com/view-vrml-online I get:
[image]
We've got some pins hovering over the board:
[image]
But it was even worse before: (https://github.com/rusefi/hellen88bmw/blob/1b9764f3ea59727ad798e16315f23c75de213dfd/boards/hellen88bmw-c/board/misc/hellen88bmw-c-3D.wrl.gz)
[image]

chuckwagoncomputing, 2023-07-15:
I think the problem here is that you're calling ext/hellen-one/bin/gha-commit.sh which doesn't make sense here, and you're calling it from the repo root rather than the directory of the board you're trying to generate gerbers for.
The result is that the files in the gerber/ directory aren't added to the commit.

rusefillc, 2023-07-15:
@chuckwagoncomputing great point about working dir for gha-commit.sh https://github.com/rusefi/rusefi-hardware/commit/972b37d1d3b69887e014d990000338ca0ea70216

rusefillc, 2023-07-15:
@chuckwagoncomputing my focus was the assert message I was assuming that it's something bad?

[image]

rusefillc, 2023-07-15:
looks like we do not care about assert message?

closed 2023-07-15 by rusefillc