rusefi/rusefi-hardware #222: Toggle digital outputs

rusefillc, 2023-07-23:
Need to expand https://github.com/rusefi/rusefi-hardware/tree/main/digital-inputs with the following logic

We need an eight second digital output cycle
* for first 1000ms toggle digital output pin 1 every 100ms the rest of the pins stay inactive
* for second 1000ms toggle digital output pin 2 every 100ms the rest of the pins stay inactive
* etc

First step is actually selecting eight pins on 144 pin STM32 Nucleo-F429ZI which we can toggle without any interference from other hardware on STM32 Nucleo-F429ZI and not yet used by existing https://github.com/rusefi/rusefi-hardware/tree/main/digital-inputs

andreika-git, 2023-08-26:
https://github.com/rusefi/rusefi-hardware/commit/9102e4aaa8623208acc5154430aec9aa0e09fa7d

rusefillc, 2023-08-30:
See #243

rusefillc, 2023-08-30:
самое полезное сейчас - провалидировать счётчики событий, полученный по CAN

```
  ADC says LOW 13@1 0.001v
               sending line=13 value=1
               sending line=13 value=0
GOOD channel 13
               Setting ADDR 13
               sending line=14 value=1
  ADC says HIGH 14@0 2.833v
               sending line=14 value=0
  ADC says LOW 14@1 0.001v
               sending line=14 value=1
               sending line=14 value=0
GOOD channel 14
               Setting ADDR 14
               sending line=15 value=1
  ADC says HIGH 15@0 2.831v
               sending line=15 value=0
  ADC says LOW 15@1 0.001v
               sending line=15 value=1
               sending line=15 value=0
GOOD channel 15
               Setting ADDR 15
               sending line=16 value=1
  ADC says HIGH 16@0 2.833v
               sending line=16 value=0
  ADC says LOW 16@1 0.001v
               sending line=16 value=1
               sending line=16 value=0
GOOD channel 16
               Setting ADDR 0
               sending line=17 value=1
  ADC says LOW 17@0 1.088v
ERROR! Cycle 17@0 FAILED! (set 1, received 0 1.088v)
               sending line=17 value=0
               sending line=17 value=1
ERROR! Cycle 17@2 FAILED! (set 1, received 0 1.087v)
               sending line=17 value=0
!!!!!!!! BAD channel 17 !!!!!!!!!!!!!!!
               Setting ADDR 1
               sending line=18 value=1
  ADC says LOW 18@0 1.087v
ERROR! Cycle 18@0 FAILED! (set 1, received 0 1.087v)
               sending line=18 value=0
               sending line=18 value=1
ERROR! Cycle 18@2 FAILED! (set 1, received 0 1.086v)
               sending line=18 value=0
!!!!!!!! BAD channel 18 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
 ++++++++++++++++ numOutputs 18
               Setting ADDR 0
               sending line=1 value=1
  ADC says HIGH 1@0 2.829v
               sending line=1 value=0
  ADC says LOW 1@1 0.002v
               sending line=1 value=1
  ADC says HIGH 2@0 2.841v
               sending line=2 value=0
  ADC says LOW 2@1 0.004v
               sending line=2 value=1
               sending line=2 value=0
GOOD channel 2
               Setting ADDR 2
               sending line=3 value=1
  ADC says HIGH 3@0 2.843v
               sending line=3 value=0
  ADC says LOW 3@1 0.003v
               sending line=3 value=1
               sending line=3 value=0
[... 1971 lines cut ...]
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 20 10 0 0 0 0
       CAN RX outputCount total=32 low=16
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 F0 22 0 3 C5 0 49 0
       CAN RX BoardStatus: BoardID=61474 numSecs=965
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1F 0 0 0 0 0 0 0
  ADC says HIGH 28@0 1.088v
               sending line=28 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 1E 1E 1E 1E 0 0 F 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 72 44 42
  ADC says LOW 28@1 0.002v
               sending line=28 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 1E 1E 1E 1E 0 0 F 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 72 44 42
               sending line=28 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 1E 1E 1E 1E 0 0 F 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 72 44 42
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 20 10 0 0 0 0
       CAN RX outputCount total=32 low=16
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 F0 22 0 3 C5 0 49 0
       CAN RX BoardStatus: BoardID=61474 numSecs=965
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1F 0 0 0 0 0 0 0
GOOD channel 28
               Setting ADDR 12
               sending line=29 value=1
```

closed 2023-09-17 by rusefillc