rusefi/rusefi-hardware #227: увеличить верхние отверстия на Breakout_154pin_284617-1-Connector

rusefillc, 2023-08-14:
мы недавно успешно удвоили нижние отверстия на Breakout_154pin_284617-1-Connector https://github.com/rusefi/rusefi-hardware/issues/218

оказалось, что на https://github.com/rusefi/hellen154hyundai немного увеличины верхние тоже.

Нужно скопировать размер и расположение верхних с https://github.com/rusefi/hellen154hyundai на Breakout_154pin_284617-1-Connector и сделать ревизию 0.6

Nadezhda-n, 2023-08-15:
Сделала. После автоматической генерации герберов внутренние слои (лишние, которых нет на плате) удалила вручную

rusefillc, 2023-08-24:
Алилуя! только версию забыли на плате поменять :)

[image]

closed 2023-08-24 by rusefillc