rusefi/rusefi-hardware #228: hellen-honda-k stim profile/pinout

rusefillc, 2023-08-19:
How do I connect https://rusefi.com/docs/pinouts/hellen/hellen-honda-k to https://rusefi.com/docs/pinouts/stim?

IAT goes to https://rusefi.com/docs/pinouts/stim/?connector=main&pin=5B
CLT goes to https://rusefi.com/docs/pinouts/stim/?connector=main&pin=6B

Where should TPS go? Where should MAP go?

andreika-git, 2023-08-20:
TPS could go to https://rusefi.com/docs/pinouts/stim/?connector=main&pin=7B
MAP could go to https://rusefi.com/docs/pinouts/stim/?connector=main&pin=8B

rusefillc, 2023-08-21:
```
2023-08-21_01_12_38_547: EngineState: BENCH_TEST Sending CAN bus0 message: ID=770000/l=8 31 0 56 50 0 0 0 0
2023-08-21_01_12_38_547: EngineState: BENCH_TEST Sending CAN bus0 message: ID=770001/l=8 1B 0 0 0 20 4F 2B 62
```

rusefillc, 2023-08-22:
[image]

rusefillc, 2023-08-22:
где-то перепутаны IAT и CLT, но где?

на фото видно, что CLT синий на хонде и синий https://rusefi.com/docs/pinouts/stim/?connector=main&pin=6B

```
Processing ID=770001/l=8 1B 0 0 0 20 2B 4F 66
 * BAD channel 5 (CLT): voltage 0.843137280 (raw 43) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.549019712 (raw 79) not in range (0.811403520..0.942982464)
Processing ID=770001/l=8 1B 0 0 0 20 2B 4F 66
 * BAD channel 5 (CLT): voltage 0.843137280 (raw 43) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.549019712 (raw 79) not in range (0.811403520..0.942982464)
```

andreika-git, 2023-08-22:
[image]

rusefillc, 2023-08-26:
@andreika-git please add "getBoardMetaOutputsCount()" and other parts of the output pattern into honda

closed 2023-09-17 by rusefillc