rusefi/rusefi-hardware #234: proteus f7 stim profile/pinout

rusefillc, 2023-08-22:
```
 * BoardStatus: BoardID=61474 numSecs=20
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770003/l=8 F0 22 0 0 14 0 3F 0
 * BoardStatus: BoardID=61474 numSecs=20
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770003/l=8 F0 22 0 0 15 0 3F 0
 * BoardStatus: BoardID=61474 numSecs=21
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770003/l=8 F0 22 0 0 15 0 3F 0
 * BoardStatus: BoardID=61474 numSecs=21
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
Processing ID=770001/l=8 0 0 0 0 0 FE FE 42
```

rusefillc, 2023-08-29:
status update: just fixed https://github.com/rusefi/rusefi/issues/5545

MAP & TPS analog works but IAT/CLT are not happy would need to be investigated

I've also wired 16 low-side and 2 high side channels hoping to compare low side vs high side processing. Unfortunately do not have 16 low sides working - have channel 9 for instance not working. open question if/what could be improved to facilitate troubleshooting - at the moment it takes too much labor to read console output. could be wiring issue could be stim issue could be anything.

```
  ADC says LOW 18@0 0.003v
ERROR! Cycle 18@0 FAILED! (set 1, received 0 0.003v)
  sending line=18 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                                             .604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..                                                                                                                                                             0.942982464)
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                                             .604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..                                                                                                                                                             0.942982464)
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                                             .604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..                                                                                                                                                             0.942982464)
ERROR! Cycle 18@2 FAILED! (set 1, received 0 0.003v)
  sending line=18 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                                             .604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..                                                                                                                                                             0.942982464)
!!!!!!!! BAD channel 18 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 12 0 0 0 0 0
 ***** got outputCount 18
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 F0 22 0 0 8D 0 49 0
 * BoardStatus: BoardID=61474 numSecs=141
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1 0 0 0 0 0 0 0
 ++++++++++++++++ numOutputs 18
Setting ADDR 0
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                                             .604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..                                                                                                                                                             0.942982464)
  ADC says HIGH 1@0 2.834v
  sending line=1 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                                             .604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..                                                                                                                                                             0.942982464)
  ADC says LOW 1@1 0.003v
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1                                                                                                                                          [... cut ...]
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 F0 22 0 0 94 0 49 0
 * BoardStatus: BoardID=61474 numSecs=148
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.333333376 (raw 68) not in range (0.811403520..0.942982464)
  sending line=1 value=0

```

andreika-git, 2023-08-29:
Proteus needs custom #define IAT_VALUE and CLT_VALUE

andreika-git, 2023-08-29:
@rusefillc Can you please re-test the IAT/CLT for Proteus?

rusefillc, 2023-08-29:
IAT is happier but not CLT

```
  sending line=6 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.968085184..2.287234080)
  ADC says HIGH 6@0 2.827v
  sending line=6 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.968085184..2.287234080)
  ADC says LOW 6@1 0.003v
  sending line=6 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.968085184..2.287234080)
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 12 0 0 0 0 0
 ***** got outputCount 18
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 F0 22 0 0 8 0 49 0
 * BoardStatus: BoardID=61474 numSecs=8
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1 0 0 0 0 0 0 0
  sending line=6 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.968085184..2.287234080)
GOOD channel 6
Setting ADDR 6
  sending line=7 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.968085184..2.287234080)
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 12 0 0 0 0 0
 ***** got outputCount 18
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 F0 22 0 0 8 0 49 0
 * BoardStatus: BoardID=61474 numSecs=8
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 1 0 0 0 0 0 0 0
  ADC says HIGH 7@0 2.827v
  sending line=7 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1B 0 0 0 20 9 44 42
 * BAD channel 5 (CLT): voltage 0.176470592 (raw 9) not in range (1.968085184..2.287234080)
  ADC says LOW 7@1 0.004v
  sending line=7 value=1

```

closed 2023-09-17 by rusefillc