rusefi/rusefi-hardware #237: Test ECU outputs using CAN

andreika-git, 2023-08-26:
https://github.com/rusefi/rusefi-hardware/commit/9102e4aaa8623208acc5154430aec9aa0e09fa7d

andreika-git, 2023-08-26:
We need to test this.

rusefillc, 2023-08-27:
@andreika-git META data was not requested by stim

I've made it broadcast https://github.com/rusefi/rusefi/commit/a872d8067dcd5f453d34addaca55241465415f6b

```
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 70 0 0 15 0 63 0
 * BoardStatus: BoardID=112 numSecs=21
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0

```

rusefillc, 2023-08-27:
```
 ++++++++++++++++ numOutputs 2
Setting ADDR 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1A 0 0 0 1E 6B 40 67
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
  HIGH 1@0 2.843v
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 70 0 1 66 0 63 0
 * BoardStatus: BoardID=112 numSecs=358
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
ERROR! Cycle 1@1 FAILED! (set 0, received 1 2.843v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
  HIGH 2@0 2.843v
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
 * BAD channel 5 (CLT): voltage 2.098039392 (raw 107) not in range (1.380596992..1.604477632)
 * BAD channel 6 (IAT): voltage 1.254902000 (raw 64) not in range (0.811403520..0.942982464)
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 70 0 1 66 0 63 0
 * BoardStatus: BoardID=112 numSecs=358
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
ERROR! Cycle 2@1 FAILED! (set 0, received 1 2.843v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
 
```

rusefillc, 2023-08-27:
```
Setting ADDR 0
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 1@0 0.001v
ERROR! Cycle 1@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 2@0 0.001v
ERROR! Cycle 2@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 0 55 0 63 0
 * BoardStatus: BoardID=11 numSecs=85
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
 ++++++++++++++++ numOutputs 2
Setting ADDR 0
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 1@0 0.002v
ERROR! Cycle 1@0 FAILED! (set 1, received 0 0.002v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 2@0 0.001v
ERROR! Cycle 2@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 0 55 0 63 0
 * BoardStatus: BoardID=11 numSecs=85
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
 ++++++++++++++++ numOutputs 2
Setting ADDR 0
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 1@0 0.001v
ERROR! Cycle 1@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 2@0 0.001v
ERROR! Cycle 2@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 0 56 0 63 0
 * BoardStatus: BoardID=11 numSecs=86
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
 ++++++++++++++++ numOutputs 2
Setting ADDR 0
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1A 0 0 0 1E 6B 40 67
  ADC says LOW 1@0 0.001v
ERROR! Cycle 1@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says LOW 2@0 0.001v
ERROR! Cycle 2@0 FAILED! (set 1, received 0 0.001v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0

```

rusefillc, 2023-08-27:
```
ERROR! Cycle 1@2 FAILED! (set 1, received 0 0.001v)
  sending line=1 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
ERROR! Cycle 1@3 FAILED! (set 0, received 1 2.851v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 1 16 0 63 0
 * BoardStatus: BoardID=11 numSecs=278
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
  ADC says LOW 2@0 0.001v
ERROR! Cycle 2@0 FAILED! (set 1, received 0 0.001v)
  sending line=2 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
  ADC says HIGH 2@1 2.852v
ERROR! Cycle 2@1 FAILED! (set 0, received 1 2.852v)
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
ERROR! Cycle 2@2 FAILED! (set 1, received 0 0.002v)
  sending line=2 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 1 16 0 63 0
 * BoardStatus: BoardID=11 numSecs=278
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
ERROR! Cycle 2@3 FAILED! (set 0, received 1 2.850v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
 ++++++++++++++++ numOutputs 2
Setting ADDR 0
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 1A 0 0 0 1E 6B 40 68
  ADC says LOW 1@0 0.002v
ERROR! Cycle 1@0 FAILED! (set 1, received 0 0.002v)
  sending line=1 value=0
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 1 16 0 63 0
 * BoardStatus: BoardID=11 numSecs=278
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
  ADC says HIGH 1@1 2.854v
ERROR! Cycle 1@1 FAILED! (set 0, received 1 2.854v)
  sending line=1 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
ERROR! Cycle 1@2 FAILED! (set 1, received 0 0.002v)
  sending line=1 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 1 17 0 63 0
 * BoardStatus: BoardID=11 numSecs=279
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
ERROR! Cycle 1@3 FAILED! (set 0, received 1 2.851v)
!!!!!!!! BAD channel 1 !!!!!!!!!!!!!!!
Setting ADDR 1
  sending line=2 value=1
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
  ADC says LOW 2@0 0.002v
ERROR! Cycle 2@0 FAILED! (set 1, received 0 0.002v)
  sending line=2 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
  ADC says HIGH 2@1 2.851v
ERROR! Cycle 2@1 FAILED! (set 0, received 1 2.851v)
  sending line=2 value=1
Processing BENCH_TEST_IO_META_INFO ID=770005/l=8 66 0 2 0 0 0 0 0
 ***** got outputCount 2
Processing BENCH_TEST_BOARD_STATUS ID=770003/l=8 0 B 0 1 17 0 63 0
 * BoardStatus: BoardID=11 numSecs=279
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 67
ERROR! Cycle 2@2 FAILED! (set 1, received 0 0.002v)
  sending line=2 value=0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 0 0 0 0 0 0 0 0
Processing BENCH_TEST_RAW_ANALOG ID=770001/l=8 19 0 0 0 1E 6B 40 68
ERROR! Cycle 2@3 FAILED! (set 0, received 1 2.852v)
!!!!!!!! BAD channel 2 !!!!!!!!!!!!!!!
 ************* SOMETHING BAD SEE ABOVE ************************

```

rusefillc, 2023-08-31:
looks to be working!

closed 2023-08-31 by rusefillc