rusefi/rusefi-hardware #255: mcz33810 test board

rusefillc, 2023-11-16:
Looking to make a test board around mcz33810 https://www.nxp.com/products/power-management/smart-switches-and-drivers/low-side-switches/automotive-engine-control-ic:MC33810

26 superseal connector any pinout

* SPI on the connector and also vias to solder wires
* GND/power
* 8 outputs 
* igbts same as https://github.com/rusefi/rusefi-hardware/issues/159

We have footprint ready https://github.com/rusefi/kicad6-libraries/commit/a723b4c5e3fc240597fa57c9205f5f1ac134d35b

rusefillc, 2023-11-16:
first part of the schematics https://github.com/andreika-git/hellen-one/issues/207#issuecomment-1336519453

rusefillc, 2023-11-17:
@Qwerty-OFF проводочек откуда куда?

Nadezhda-n, 2023-11-21:
> first part of the schematics [andreika-git/hellen-one#207 (comment)](https://github.com/andreika-git/hellen-one/issues/207#issuecomment-1336519453)

[image]
Эта часть заменяется на https://github.com/rusefi/rusefi-hardware/issues/159 ?
Или на другие какие-то цепи ставить (IGN1/D_IGN1)?

closed 2023-11-21 by rusefillc