rusefi/rusefi-hardware #260: two issues with Breakout-24pin

rusefillc, 2024-02-25:
breakout-boards/Breakout-24pin

1) крайние шесть пинов побольше, там отверстия под провод нужно 1.8мм диаметра - это пины A1 B1 C1 A8 B8 C8
https://rusefi.com/docs/pinouts/hellen/alphax-2chan/

2) нужно уменьшить плату над разъёмом - сейчас он не надевается правильно

Nadezhda-n, 2024-02-26:
> breakout-boards/Breakout-24pin
> 
> 1. крайние шесть пинов побольше, там отверстия под провод нужно 1.8мм диаметра - это пины A1 B1 C1 A8 B8 C8
>    https://rusefi.com/docs/pinouts/hellen/alphax-2chan/
> 2. нужно уменьшить плату над разъёмом - сейчас он не надевается правильно

@rusefillc насколько уменьшить? Как он надевается?

Nadezhda-n, 2024-02-28:
[image]

rusefillc, 2024-03-04:
[image]

Nadezhda-n, 2024-03-05:
> @Nadezhda-n need 24pins_In_3conn_0.2_gerbers.zip file ideally with 0.2 silk

на шелке все было ок, файлик с нужным названием добавила