rusefi/rusefi-hardware #261: revive tle9104

rusefillc, 2024-03-05:


Nadezhda-n, 2024-03-13:
@rusefillc 
Компоненты для сквозного монтажа тоже должны иметь номер LCSC?

rusefillc, 2024-03-13:
@Nadezhda-n давай попробуем без стресса

Nadezhda-n, 2024-03-13:
> @Nadezhda-n давай попробуем без стресса

Тогда вроде обновила https://github.com/rusefi/rusefi-hardware/blob/main/tle9104-breakout/export/tle9104-breakout-bom.csv
Посмотри, все ли я поняла верно?

rusefillc, 2024-03-28:
@Nadezhda-n не вижу placement файла?

Nadezhda-n, 2024-03-29:
> @Nadezhda-n не вижу placement файла?

потерялся...( но уже найден и в нужной папке!

rusefillc, 2024-03-29:
@Nadezhda-n у тебя есть VPN или другой способ протестировать возмозность заказа? что-то у меня не принимает файлы ругается на формат

Nadezhda-n, 2024-03-29:
> @Nadezhda-n у тебя есть VPN или другой способ протестировать возмозность заказа? что-то у меня не принимает файлы ругается на формат

vpn есть, но то ли для Румынии не делают монтаж, то ли что-то не работает... 
Но я помню, что просто файлы не работали, я их по старому примеру отредактировала. Проверь еще раз, пожалуйста

closed 2024-09-29 by rusefillc