rusefi/rusefi-hardware #266: GDI4: spark from R71

rusefillc, 2024-05-15:
R71 has acted as fuse in some human error (?) scenario

[image]

NOx-z, 2024-05-15:
The board is a "Rev A", and appears to be connected to Injector1_P. Prior to this blowing I was doing short circuit tests on INJ1 by twisting the wires together (no injector). All of the tests did a single spike for 2.1uS, likely the number of clock cycles the PT2001 is required to have from START to Erroring out. For these tests I also re-wrote (commented out) the microcode to remove  "boost0_err" which would be called in an under-time threshold.

Upon doing a number of tests I reverted the code and the TBoostMin (from 0 or 1) back to 100.. Amperage set to a rather massive 24A
This resistor blew on first test click with real injector.
Bank1 (INJ1/INJ2) failed to click.
Bank2 (INJ3/INJ4) successfully clicked the same injector.

Looking into the datasheets these appear to be rated at 1/8 Watt.. this has been plenty for every single situation except for this one.. and they seem to be doing a fine job at protecting further..

I need to verify the microcode and configuration was all set correctly, replace this resistor, and try to do a normal click again to verify nothing else blew.

.. Is it worth pursuing the short circuit test further? One goal was simply to see capabilities of the board after pushing it to a moderate degree with these race injectors, and I was curious about upper limit elsewhere..

NOx-z, 2024-05-15:
I removed the open resistor, and added a removeable jumper, and am now getting clicks again on INJ1.
I actually needed to regenerate microcode and reupload for full sequence (it was only doing Boost, not peak nor hold), so this is likely pointing to some issue in there along with my tests.
 
I dont feel like frying anything more important (though I have many spares) with more tests..

What was the original purpose of these 0R ?
do we like that it may fuse (under extreme circumstances) ?