rusefi/rusefi-hardware #280: предложение по улучшению схемы GDI4

romioRN, 2024-10-22:
Предлагаю спроектировать подсоединение проводки к разъемам ЭБУ таким образом, чтобы проводка двигателя не пересекалась с проводкой кузова для того, чтобы упростить монтаж/демонтаж и уменьшить количество соединений в деталях жгута проводки.

rusefillc, 2024-10-28:
https://github.com/romioRN/fw-custom-Cat-GDI/tree/main/%D0%91%D0%BB%D0%BE%D0%BA%20%D1%83%D0%BF%D1%80%D0%BE%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D1%8F%20%D0%BF%D1%80%D1%8F%D0%BC%D1%8B%D0%BC%20%D0%B2%D0%BF%D1%80%D1%8B%D1%81%D0%BA%D0%BE%D0%BC/%D0%BF%D0%BB%D0%B0%D1%82%D0%B0%20GDI/%D0%9F%D0%BB%D0%B0%D1%82%D0%B0%20GDI

rusefillc, 2024-10-28:
[image]