rusefi/rusefi-hardware #284: GDI-4ch to broadcast GM fuel pressure sensor data over CAN

rusefillc, 2024-11-15:
We need the GM fuel sensor decoder https://github.com/rusefi/rusefi-hardware/tree/main/SENT-box/firmware into https://github.com/rusefi/rusefi-hardware/tree/main/GDI-4ch/firmware please

dron0gus, 2024-12-14:
step 1 https://github.com/rusefi/rusefi-hardware/pull/286

dron0gus, 2024-12-14:
GDI-4ch rev d.
IN_DIN4 is used for SENT.
C101 removed
R104 replaced with 4.7K.
GM fuel pressure sensor:
```
---- SENT input 0 ----
Unit time 62 timer ticks
Pause pulse detected No
Total pulses 1215989
Last valid fast msg Status 0xC Sig0 0x0C5 Sig1 0x0CB
Slow channels:
 ID 16: 756
 ID 22: 755
HW overflows 0
Pause pulses 0
Restarts 0
Interval errors 0 short, 1 long
Total frames 135141 with CRC error 0 (0%)
Total slow channel messages 7507 with crc6 errors 0 (0%)
Sync errors 0
--------------------
```

dron0gus, 2024-12-14:
Temperature is transmitted in two slow channels: 16 and 22. Convertion is still not clear:
Temperature is close to zero:
```
 ID 22: 565
 ID 16: 566
```

Below zero
```
 ID 22: 527
 ID 16: 528

 ID 22: 538
 ID 16: 538
```

Room temperature (20..22 C)
```
 ID 22: 689
 ID 16: 689
```

Just boiled water
```
 ID 22: 1220
 ID 16: 1221

 ID 22: 1202
 ID 16: 1203
```

Cooled a bit
```
 ID 22: 1151
 ID 16: 1152
```